NETMF/llilum #21: Beware Cygwin before building LLVM!

maiorfi, 2015-07-30:
In order to build LLVM cygwin (if installed) directory should be renamed (e.g. to "_cygwin64"), or CMake configuring phase won't work with many "file missing" errors.

smaillet-ms, 2015-08-10:
We don't use Cygwin to build LLVM, we use CMake to generate Visual studio solutions and projects and use msbuild to build it all. see the [BuildLlvmWithVS](https://github.com/NETMF/llilum/blob/dev/Zelig/Zelig/CompileTime/Llvm.NET/LibLLVM/BuildLlvmWithVS.cmd) build script for details

maiorfi, 2015-08-13:
Indeed issues I had running CMake scripts while having cygwin on my PATH were absolutely unexpected. It seems that CMake turns to *-nix like file paths (e.g. /cygdrive/c/myfolder/... instead of c:\myfolder\...) depending on cygwin availability on current session.

closed 2015-08-13 by maiorfi

smaillet-ms, 2015-08-13:
Interesting, are you aware of any means to detect that situation that we could add to the  [BuildLlvmWithVS](https://github.com/NETMF/llilum/blob/dev/Zelig/Zelig/CompileTime/Llvm.NET/LibLLVM/BuildLlvmWithVS.cmd) script to at least warn users?