NETMF/llilum #23: What about using LLVMSharp instead of LLVM.NET?

maiorfi, 2015-08-05:
LLVM.NET latest commit is one year old, while LLVMSharp is alive and kicking (and developed by Mukul Sabharwal, currently employed in Microsoft).

lt72, 2015-08-10:
I am actually in contact with Mukul.  We did not define any common roadmap yet. LLVM.NET though for us is just translation from our proprietary IR to bitcode, not a critical part of the system, so the value is about engineering alone.

smaillet-ms, 2015-08-10:
LLVM.NET that we are using is completely new - not sure what you are referring to with it having commits that are a year old, at present LLVM.NET has more functionality than LLVMSharp currently does - including a full and robust OO model on top of the interop to "C" with extended support for the debugging. This was migrated into C# from a previous C++/CLI wrapper due to problems dealing with the use of C++11 constructs in LLVM not supported by the C++/CLI compiler that made it harder to deal with using C++/CLI as the wrapping language.

lt72, 2015-08-11:
I guess Lorenzo (Maiorfi) is referring to some other OSS project called LLVM.NET on github... is that correct?
Lorenzo, Steve wrote the one we are using from scratch. We initially went C++/CLI, but current solution has much better debug -ability and maintainability properties. Next thing to add is more DWARF support.

maiorfi, 2015-08-13:
I'm sorry, I definitely referred to wrong LLVM.NET project :(

closed 2015-08-13 by maiorfi

smaillet-ms, 2015-08-13:
No worries, turns out there are actually a few LLVM.NET variant projects at some level or another of functionality for various LLVM versions. Unfortunately none of them were complete or current enough for what we are doing here. Once we've finished support for proper full debug information and think that LLVM.NET is solid it could published it in a separate repo to continue on independently.