NETMF/llilum #24: First steps with SDK debugging

maiorfi, 2015-08-18:
I just started playing a little with SDK generation and build+deploy+debug cycle.

Despite halting debugger actually breaks GDB, it seems that breakpoints are ignored (for example in Main()) and sources shown while halted aren't correct (I usually "find" myself inside Dictionary class sources :smile: !).

How could I be sure debug symbols are correctly loaded?

Thanks!

lt72, 2015-08-19:
No locals or args watching or inspection is available yet. Just breakpoints. We think we can get inspection done by end of this month though. 
Visual Studio does not offer great diagnostics on ELF symbols yet. To make sure your symbols are correct, I would lean on GDB command line. Symbols are in the ELF file, so I would try and deploy the ELF file generated by Visual Studio to the board using a sequence of commands from the DOS shell and GDB. Assuming that your executable image is mbed_simple.bin, the ELF is mbed_simpl.elf, and the board shows up as drive E:, I would do the following:
1) open a DOS shell
2) ```copy mbed_simle.elf E:\.```
3) reset board 
4) ```start /I pyocd_win.exe```
5) arm-none-eabi-gdb.exe" mbed_simple.elf  -ex "target remote :3333" 

(steps 4) and 5) assume gdb executable and pyocd_win.exe are in the path of course) 

now, from the gdb prompt, I would continue with

6) ```monitor halt reset```
7) ```ni```
8) ```i r```

(steps 6), 7) and 8) respectively reset the processor, which takes effect at the next stepping action, and then you can inspect the register and the PC register should be at <Reset_Handler>, which means you are restarting from scratch. 

9) b main
10) cont

at step 10) you should it the breakpoint on main, which is actually aliased to a Zelig method for bootstrapping (```Bootstrap::Initailization```)

if the above does not work, I would still try the following:

from the gdb prompt, do ```load mbed_simple.elf```. This will push the BIN inside the ELF again to the board, and also reload the symbols. Also you can try ```file mbed_simple.elf```, which just reloads the symbols.

After all of the above, you will know whether your ELF is good or not. If you still cannot hit _main_ correctly, then it means the ELF is not good, and something went wrong while building. 

If you can hit main, then VS is not deploying correctly. 

Side note: we are fixing the annoying issue where you need to kill pyOCD_win.exe manually each time...

maiorfi, 2015-08-19:
Ok, thanks.

Things go a little better using tools like VisualGDB, though I still can't understand how a native debugger like gdb could ever be able to show variable data coming from .NET "transpiled" code. I suspect that "managed" variable inspecting (e.g. browsing properties and/or non primitive types data) won't be available: right?

closed 2015-08-19 by maiorfi

lt72, 2015-08-20:
As a matter of fact we are working on that already. GDB indeed cannot understand objects, but we can easily use GDB to read memory. We can then leverage the information in the PDBs to match the memory with the object layout on the host side, and then provide Visual Studio with the correct information. We are doing this for NETMF already, this is not different in principle. 
Native types instead can be handled as DWARF metadata through LLVM, and will work natively from GDB.