NETMF/llilum #25: Error when building native application.

vgolovanov, 2015-09-22:
Hi.
I have a problem when compiling sample application.
During compiling process, zeling compiler stop working with message "Zelig Compiler has stop working".
From log messages, i see the error: 

"Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
     at Llvm.NET.LLVMNative.DisposeModule(LLVMModuleRef M)
     at Llvm.NET.Module.Dispose(Boolean disposing)
     at Llvm.NET.Module.Finalize()
"
Any help is appreciated. ) 
Thank you.

lt72, 2015-09-22:
Thank you for reporting this, It is a known issue that does not break the code generation. It only happens at the end of the LLVM translation when we shut down the LLVM context. We have a race condition for which we shut down the LLVM context twice. At that point all code is already generated and you can move onto the next steps. 
We are fixing this issue with the next push.

vgolovanov, 2015-09-23:
Thank you for your prompt reply. I have 2 more questions, not related to that problem. As i understand LLilum supports full c#, not a subset as NetMf ? Also i can't find implementations for SPI and I2C buses, is not implement yet?

Skobovm, 2015-09-25:
Hi,
Sorry about the late reply! SPI and I2C are both ready to go, and will be made available in the very near future. As for .NET framework, since the target devices are even more resource constrained than the NETMF devices, Llilum will support libraries that are applicable to embedded and IoT development.

commit bd03e8dd (2015-10-01): Added support for following types or namespaces:

- System.Threading
- System.Threading.Tasks
- Windows.System.Threading (UWP threading)
- Windows.Devices.I2c
- Windows.Devices.Spi

Also:
+ Support for Context switch for M3 and M4 (w and w/o floating point, no lazy stacking yet)
+ Robustified SDK (still needs plenty of work)
+ Revised LLVM translation
+ Revised and refined HW model
+ Laid foundations for ref counting GC

fixes #25
fixes #17
fixes #12
fixes #11
fixes #9
closed 2015-10-01 by lt72