NETMF/llilum #27: Support Python

lt72, 2015-10-03:
How about supporting IronPython assemblies AOT compilation for LLILUM devices?

IronManMark20, 2015-12-22:
While I'd love to see this, the issues I could see with IronPython is that a) it is dynamic (I don't know if you are supporting dynamic languages) and b) it doesn't run on .Net Core (which is what you are targeting?)

So, I am unsure of how you would do this.

lt72, 2016-01-06:
Definitely we would not be able to support any feature of a dynamic language per se, but we would restrict support to the non-dynamic portion. That may be a deal breaker... for sure, but not necessarily.
The way to support Python and/or JS (TypeScript) is to implement the runtime functionalities that the Python (JS/TypeScrpt) compiler lean upon.  
Not different than VB.NET support for .Net Micro Framework.

The issue here is: is support for a dynamic language interesting it has to cut the dynamic portion of it?