NETMF/llilum #30: Scheduler extensions and specializations

lt72, 2015-10-04:
This issue is the placeholder for any discussion related to thread scheduling and management. Please provide your ideas, or any requirement you may have.

cpfister, 2015-10-04:
For some kinds of real-time applications, I'd like to be able to build a scheduler for "time-triggered" tasks where the only kind of event to react on is the advance of the system clock. This often makes it easier to ensure correctness of such a system, as the application itself is in control of I/O, rather than having to react on interrupts no matter in what state the system currently is.

This may be a bit a special requirement though, so rather than being a standard library feature it probably should be something that we can develop ourselves when needed.

cw2, 2015-10-07:
Add support for tickless mode.