NETMF/llilum #34: Target compile error.

igvas, 2015-10-06:
I am try to compile K64F target and receive linkage error:

Link:
D:\mydocs\gitHub\llilum\Scripts\SDKDrop\mbed\TARGET_K64F/TOOLCHAIN_GCC_ARM/mbed_overrides.o: In function `NMI_Handler':
mbed_overrides.c:(.text.NMI_Handler+0x0): multiple definition of `NMI_Handler'  ARM\Debug\SimpleTestProgram.o:D:\mydocs\gitHub\llilum\Zelig\Zelig\RunTime\Zelig\Kernel\HardwareModel\TargetPlatform\ARMv7/ProcessorARMv7M_ContextSwitch.cs:651: first defined here
collect2.exe : error : ld returned 1 exit status

Can anyone help me?

lt72, 2015-10-06:
You need to change the configuration file for the FrontEnd project. 
Go to the project
<repo>\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd 
and change the debug properties to target file mbed_simple_K64F.FrontEndConfig. 
Then change the #define at the top of the file <repo>\llilum\Zelig\Zelig\Test\mbed\simple\Program.cs 
from this: 

#define LPC1768
//#define K64F

to this:

//#define LPC1768
#define K64F

Skobovm, 2015-10-06:
I believe you may be building the SDK-based test project.
If you are building the Native project under Test/mbed/SDK then do the following:
1. Go to the Native project properties, click the Llilum Configuration tab, and select the K64F board
2. In that tab, put $(MSBuildProjectDirectory)\mbed_simple_K64F.FrontEndConfig as the Llilum Config
3. Change the the #define statements as stated above

igvas, 2015-10-06:
I have already done all steps. Error raises after that. NMI_Handler exists in both files: mbed_overrides.o (i guess  precompiled from mbed_overrides.c) and in ProcessorARMv7_ContextSwitch.cs. I will check all twice again.

Skobovm, 2015-10-06:
I just pulled a clean repo and this happens for me as well. Thanks for pointing this out! We will definitely investigate this.

lt72, 2015-10-07:
Hi igvas,
I verified that the problem exists for the SDK -style project in the solution. In reality all you need to do is to just get rid of mbed_overrides.o, we do not need it any longer. That requires modifying the SDK though, so we will do that for you ASAP.
In the meantime, you can just comment out the NMI_Handler function in file 
```<repo>\Zelig\Zelig\RunTime\Zelig\Kernel\HardwareModel\TargetPlatform\ARMv7_VFP\ProcessorARMv7M_VFP_ContextSwitch.cs```, line 913, and also the same code in file ```<repo>\Zelig\Zelig\RunTime\Zelig\Kernel\HardwareModel\TargetPlatform\ARMv7\ProcessorARMv7M_ContextSwitch.cs```. 

While investigating, I found out Timers are still unreliable on K64F. 
If you look in Progrqam.cs (file ```<repo>\Zelig\Zelig\Test\mbed\SimpleSDK\Managed\Program.cs```) ou will see that there is come commented out that looks like the one below:

```
var solitaryAlerter = new System.Threading.Timer( ( obj ) => { ((AutoResetEvent)obj).Set(); }, ev, 2000, 5000 );
Microsoft.CortexM3OnMBED.Drivers.SystemTimer.Timer tm = Microsoft.CortexM3OnMBED.Drivers.SystemTimer.Instance.CreateTimer( ( timer, currentTime ) => { ev.Set( ); tm.RelativeTimeout = 5 * 1000* 1000; } );
tm.RelativeTimeout = 5 * 1000 * 1000;
```
that is indeed the low level timer, which should work fine, if you need a timer. 

Thank you for your patience by the way!

P.S.: the makefile project in <repo>\Zelig\LLVM2IR_results\mbed\simple works much better at this stage... but we are in the middle of a push to make the SDK much more reliable that it is now, so you will see daily improvements.

lt72, 2015-10-10:
fixed with 9f6273a

lt72, 2015-10-10:
Timers fixed as well

closed 2015-10-10 by lt72