NETMF/llilum #42: Some K64F issues

roceh, 2015-10-10:
I know you just committed a lot of these changes, but here are the few issues I had:

Some of the stubs are not defined for the K64F as far as I can tell, e.g. 

CUSTOM_STUB_DebuggerConnected
CUSTOM_STUB_SCB__get_HFSR
CUSTOM_STUB_SCB__get_CFSR
CUSTOM_STUB_SCB__get_BFAR

I also had an issue where the processor context was failing a cast on the thread context and breaking, it looked like the thread context was the one in ProcessorARMv7M_VFP but the runtime was casting to the ProcessorARMv7M.Context class. Changing the processor on the K64F to derive from ProcessorARMv7M instead of ProcessorARMv7M_VFP got it working for me.

Noticed some debugging issues, breakpoints sometimes work, sometimes not or behave oddly (i.e. break wrong place). The callstack listed the line numbers correctly but double clicking to go to them jumped to random locations. 

In the end I did get the little blinky test app working via the SDK, and it is all very cool =) looking forward to alpha. Also if your going to support more mbed boards the STM32F7 DISCO board with touch screen LCD would be very cool to see.

lt72, 2015-10-12:
Hi, I propagated now the implementation for the methods above. It was already available for the makefile project in Test directory, but we did not apply that to the SDK template. Please check 
https://github.com/NETMF/llilum/blob/dev/Zelig/Zelig/Test/mbed/SimpleSDK/Native/InteropExports_debug.cpp 
or
https://github.com/NETMF/llilum/blob/dev/VisualStudio/LlilumProjectType/LlilumApplication/LlilumApplication.ProjectTemplate/LlilumNative/InteropExports_debug.cpp 

we will remove the duplication shortly.

The cast issue could be related to building for LPC1768 while activating the #define for K64F in Program.cs, or viceversa. 
You may want to check that both your native C++ project setting (Native ==> Properties ==> Lillum Configuration) and your #define in Program.cs point to K64F or LPC1768.

lt72, 2015-10-12:
On another note: we are working on fixing debugging as we speak. ETA is 10/23 to have everything working smoothly.
And yes, we will add support for STM32Fxxx DISCO. Not sure if F4 or F7

roceh, 2015-10-12:
Ah thanks for the info I'll grab them from the test app. With the context issue I think my defines are ok, the cast issue might be due to ProcessorARMv7M_VFP class redefining the embedded Context object but inheriting from Processor.Context - I think it should be inherited from ProcessorARMv7M.Context instead.

lt72, 2015-10-13:
K64f is an M4 and you need to use the _VFP variant of the processor type to account for FP registers stacking/unstacking during context switch. Using the non- _VFP variant will basically treat the M4 processor like an M3, which is fine if you do no FP operation, so that the FP registers are never dirty and need no stacking/unstacking.
The compilation setup depends on the #define in the managed app, but also on the set of device drivers you select. Such drivers are provided to the compiler (directly or indirectly) by selecting the FrontEnd config file in the project properties.
Some mix up is likely happening at that level.  I will try and repro it if you give me more details:
1) are you building with the makefile project or are you using the SDK?
2) if you are using the SDK, did you change the Llilum native project properties to target K64F? (see attachment)
3) could you please send the call stack where the cast happens?
[image]

roceh, 2015-10-13:
Tracked down the problem due to SDK build not setting -mfloat-abi=hard so __SOFTFP__ is set and __FPU_USED is never set which in turn means SVC_Handler_Zelig is called instead of SVC_Handler_Zelig_VFP_NoFPContext which causes the ProcessorARMv7M class try to cast a ProcessorARMv7M_VFP context to a ProcessorARMv7M.Context. Setting -mfloat-abi=hard fixes the issue.

lt72, 2015-10-14:
Great! Thank you for the finding.  Mikhail, can you please make the change directly on the LLILUM repo and close the issue after verifying?

Skobovm, 2015-10-14:
This issue has been fixed with the latest commit. In order to apply the fix globally, you will have to pull, and re-install the SDK. Otherwise, you can make the change manually for every new Llilum project, since you already know what the fix is.

closed 2015-10-14 by Skobovm

roceh, 2015-10-17:
As a follow up I found -mfpu=fpv4-sp-d16 -mfloat-abi=softfp achieves the same and is less linker hassle (its what you currently have in your simple test makefile)