NETMF/llilum #44: K64F debugging error

igvas, 2015-10-14:
I try to debug K64F platform. Deployment is ok, but debug session do not starts.
1>------ Deploy started: Project: Native, Configuration: Debug ARM ------
1>INFO:root:new board id detected: 02400201C5271E6438D9E3DC
1>INFO:root:board allows 5 concurrent packets
1>INFO:root:DAP SWD MODE initialised
1>INFO:root:IDCODE: 0x2BA01477
1>INFO:root:K64F not in secure state
1>INFO:root:6 hardware breakpoints, 4 literal comparators
1>INFO:root:CPU core is Cortex-M4
1>INFO:root:FPU present
1>INFO:root:4 hardware watchpoints
1>
1>[                    ]   0%
1>[                    ]   3%
1>[=                   ]   6%
1>[=                   ]   9%
1>[==                  ]  12%
1>[==                  ]  15%
1>[===                 ]  18%
1>[====                ]  21%
1>[====                ]  24%
1>[=====               ]  26%
1>[=====               ]  29%
1>[======              ]  32%
1>[=======             ]  35%
1>[=======             ]  38%
1>[========            ]  41%
1>[========            ]  44%
1>[=========           ]  47%
1>[==========          ]  50%
1>[==========          ]  53%
1>[===========         ]  56%
1>[===========         ]  59%
1>[============        ]  62%
1>[============        ]  65%
1>[=============       ]  68%
1>[==============      ]  71%
1>[==============      ]  74%
1>[===============     ]  76%
1>[===============     ]  79%
1>[================    ]  82%
1>[=================   ]  85%
1>[=================   ]  88%
1>[==================  ]  91%
1>[==================  ]  94%
1>[====================] 100%
1>WARNING: file already exists but should not: 
...\Temp\_MEI86002\include\pyconfig.h
========== Deploy: 1 succeeded, 0 failed, 0 skipped ==========

[image]

Skobovm, 2015-10-14:
Can you please go to Visual Studio, open the Command Window from View->Other Windows -> Command Window and try typing in Debug.MIDebug... If you don't have Debug.MIDebugLaunch and Debug.MIDebugExec come up as options, you are missing the VS GDB tooling.  

[image]  

This can be acquired by getting the prerequisites from here: https://github.com/Microsoft/MIEngine/ If the issue persists after following these steps, please let us know

igvas, 2015-10-14:
I have get and compile MIEngine. After building process i get Microsoft.MIDebugPackage.DoNotInstall.vsix in bin\Debug output folder. I have installed package despite of DoNotInstall in file name.

Situation do not changed. Even if i try run it from command line:
>Debug.MIDebugLaunch /OptionsFile:d:\mydocs\gitHub\llilum\Zelig\LLVM2IR_results\mbed\simple\DebugOptions.xml /Executable:d:\mydocs\llilum\work\LlilumApplication1\ARM\Debug\LlilumApplication1.elf
Unable to start program 'd:\mydocs\llilum\work\LlilumApplication1\ARM\Debug\LlilumApplication1.elf'.
Не удается найти указанный файл.

Last error from Russian : file not found

Skobovm, 2015-10-14:
This is what the prerequisites section has on the MIDebugger page:
"MIEngine requires Visual Studio 2015 with the following features installed:

Programming Languages -> Visual C++ -> Common Tools for Visual C++
Cross Platform Mobile Development -> Visual C++ Mobile Development
Cross Platform Mobile Development -> Microsoft Visual Studio Emulator for Android
Common Tools -> Visual Studio Extensibility Tools"  

If you can, in VS go to Tools -> Extensions and Updates, and uninstall the VSIX that was installed.

Other things to try:
1. Verify that the .elf file is in that exact location. Does the error window or output window in VS tell you which file is missing? Have you built successfully? 
2. DebugOptions.xml references a debug.bat file, try making the path to it absolute
    2.1. In the debug.bat file, make sure that your GCC_BIN variable is set correctly
3. Try changing the Board property of the native project to LPC1768 and see if the outcome changes. The program will not work, however it would narrow down the problem
4. After trying to debug from VS, in the Native project folder, you will have a tmp_debug.bat file. Make sure that file exists, and make sure that the paths in it are correct
5. Try debugging from VS again, go to the Output window, for the "Show output from" dropdown, select Debug, and see if that has any useful information
6. Copy the .bin file, not the .elf to the K64F drive, and see if it runs, to ensure your device is working properly

igvas, 2015-11-12:
Sorry for delay with answer. Now all works fine. The reason is: i've missed required installation:(https://github.com/NETMF/llilum/wiki/SDK-User-Guide) **Visual Studio Project System Extensibility Tools**. Thanks for helping.

closed 2015-11-12 by igvas