NETMF/llilum #46: SDK Build error

mv64, 2015-10-19:
I have tried to build SDK with CreateSDKDrop.bat, but I got the following error :
E:\src\llilum\Zelig\Zelig\Test\mbed\SimpleSDK\Native\Native.vcxproj(26,5): error MSB4019: The imported project "E:\src\llilum\Zelig\Zelig\Test\mbed\SimpleSDK\Native\Platforms\Llilum_LPC1768.props" was not found. Confirm that the path inthe <Import> declaration is correct, and that the file exists on disk.
[image]

I have also noticed lot of errors when building Zeil image with FrontEnd project - Phase: DetectNonImplementedInternalCalls:
[image]

 129,724 [   0,891]: Phase: DetectNonImplementedInternalCalls
ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_DebuggerConnected()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB__get_HFSR()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB__get_CFSR()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB__get_BFAR()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__get_CONTROL()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__get_IPSR()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__get_PSP()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__set_PSP(uint)

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__get_PRIMASK()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__get_BASEPRI()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_SCB__set_BASEPRI(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB_set_CCR(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB_SHCRS_EnableSystemHandler(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB_ICSR_RaiseSystemException(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_RaiseSupervisorCallForLongJump()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_RaiseSupervisorCallForStartThreads()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_RaiseSupervisorCallForRetireThread()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SetExcReturn(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CUSTOM_STUB_SCB_SCR_SetSystemControlRegister(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::CMSIS_STUB_POWER_WaitForEvent()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::Breakpoint(uint)

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M::Nop()

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M_VFP::CUSTOM_STUB_SCB__get_FPCCR()

ERROR: missing implementation for void Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M_VFP::CUSTOM_STUB_SCB__set_FPCCR(uint)

ERROR: missing implementation for uint Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M_VFP::CUSTOM_STUB_GetFPContextFlag()

ERROR: missing implementation for void Microsoft.Zelig.Support.mbed.Utilities::Breakpoint(uint)

ERROR: missing implementation for void Microsoft.Zelig.Support.mbed.Utilities::DebugLog0(char*,uint)

ERROR: missing implementation for void Microsoft.Zelig.Support.mbed.Utilities::DebugLog1(char*,uint,int)

ERROR: missing implementation for byte* Microsoft.Zelig.Support.mbed.Memory::malloc(uint)

ERROR: missing implementation for byte* Microsoft.Zelig.Support.mbed.Memory::free(byte*)

ERROR: missing implementation for uint Microsoft.Zelig.Support.mbed.UsTicker::us_ticker_read()

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.Drivers.SystemTimer::tmp_sys_timer_init()

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.Drivers.SystemTimer::tmp_timer_event_alloc(Microsoft.CortexM3OnMBED.Drivers.SystemTimer.TimerEve
ntImpl**)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.Drivers.SystemTimer::tmp_sys_timer_insert_event(Microsoft.CortexM3OnMBED.Drivers.SystemTimer.Tim
erEventImpl*,uint)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.Drivers.SystemTimer::tmp_sys_timer_remove_event(Microsoft.CortexM3OnMBED.Drivers.SystemTimer.Tim
erEventImpl*)

ERROR: missing implementation for uint Microsoft.CortexM3OnMBED.Drivers.SystemTimer::tmp_sys_timer_read()

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.GpioPinMbed::tmp_gpio_write(Microsoft.CortexM3OnMBED.HardwareModel.GPIOimpl*,int)

ERROR: missing implementation for int Microsoft.CortexM3OnMBED.HardwareModel.GpioPinMbed::tmp_gpio_alloc(Microsoft.CortexM3OnMBED.HardwareModel.GPIOimpl**)

ERROR: missing implementation for int Microsoft.CortexM3OnMBED.HardwareModel.GpioPinMbed::tmp_gpio_init(Microsoft.CortexM3OnMBED.HardwareModel.GPIOimpl*,int)

ERROR: missing implementation for int Microsoft.CortexM3OnMBED.HardwareModel.GpioPinMbed::tmp_gpio_free(Microsoft.CortexM3OnMBED.HardwareModel.GPIOimpl*)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.GpioPinMbed::tmp_gpio_mode(Microsoft.CortexM3OnMBED.HardwareModel.GPIOimpl*,Micros
oft.CortexM3OnMBED.HardwareModel.PinMode)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.GpioPinMbed::tmp_gpio_dir(Microsoft.CortexM3OnMBED.HardwareModel.GPIOimpl*,Microso
ft.CortexM3OnMBED.HardwareModel.PinDirection)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.SpiChannel::tmp_spi_alloc(Microsoft.CortexM3OnMBED.HardwareModel.SpiImpl**)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.SpiChannel::tmp_spi_init(Microsoft.CortexM3OnMBED.HardwareModel.SpiImpl*,int,int,i
nt,int)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.SpiChannel::tmp_spi_format(Microsoft.CortexM3OnMBED.HardwareModel.SpiImpl*,int,int
,int)

ERROR: missing implementation for void Microsoft.CortexM3OnMBED.HardwareModel.SpiChannel::tmp_spi_frequency(Microsoft.CortexM3OnMBED.HardwareModel.SpiImpl*,int)

ERROR: missing implementation for void Microsoft.DeviceModels.Chipset.CortexM3.NVIC::CMSIS_STUB_NVIC_SetPriorityGrouping(uint)

ERROR: missing implementation for void Microsoft.DeviceModels.Chipset.CortexM3.NVIC::CMSIS_STUB_NVIC_SetPriority(Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.Pr
ocessorARMv7M.IRQn_Type,uint)

ERROR: missing implementation for uint Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_SysTick_GetCTRL()

ERROR: missing implementation for uint Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_SysTick_GetVAL()

ERROR: missing implementation for uint Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_SysTick_GetCALIB()

ERROR: missing implementation for void Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_SysTick_SetCTRL(uint)

ERROR: missing implementation for void Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_SysTick_SetLOAD(uint)

ERROR: missing implementation for void Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_SysTick_SetVAL(uint)

ERROR: missing implementation for uint Microsoft.DeviceModels.Chipset.CortexM3.SysTick::CMSIS_STUB_CLOCK__GetSystemCoreClock()

lt72, 2015-10-19:
Hi The ERROR message should actually be a warning. Those methods are missing because they are implemented in C, and they are internal calls, so that is by design.
As for the .props file, you should not be trying and using the SDK project unless they LLILUM SDK .vsix is installed with you Visual Studio 2015. 
Please either disable the native and SimpleTestProgram project under the SDK folder or build and install the LLILUM SDK (instructions [here](https://github.com/NETMF/llilum/wiki/SDK-User-Guide)) . If you choose to disable you can still build, run and test using the [makefile project](https://github.com/NETMF/llilum/wiki/demo).

Skobovm, 2015-10-19:
Hi,

There is actually an issue with the .props file, where it wants to have the SDK installed before creating the SDK. Please follow this temporary workaround:  
  
1. Open the Zelig solution in <repo>\Zelig\Zelig\Zelig.sln  
2. Unload the Native project and SimpleTestProgram project (right click on project -> Unload Project)  
3. Rebuild the solution  
4. Edit CreateSDKDrop.bat and remove the 5 lines under "echo Building Zelig compiler..."  
5. Save the file, and then follow the standard SDK instructions from the beginning.

smaillet-ms, 2015-10-29:
I'd argue the errors should be eliminated and not even treated as a warning (I have a fix for that). These are all imported methods from "C" for which there never will be a managed body - warning about something that is intended with nothing the user can do to change is just annoying, in my view.

lt72, 2015-10-30:
the errors have been eliminated already with change d0fb0a973b53438def4a7cf233afdcb2801b72e3

lt72, 2015-10-30:
We will eventually fix the SDK installation issue

closed 2015-11-30 by Skobovm