NETMF/llilum #50: Buiding Error

weixiongmei, 2015-10-26:
Have been trying to buid this project for couple days, but still can't get it works, always showing errors like:

can't open the file: "llvm/Support/Datatypes.h": No such file or directory
.....
can't open the file: "llvm/Config/llvm-config.h": No such file or directory
.....

Have no idea where i can get those header files, i install the LLVM 3.7.0 binary on my windows 10 already. Please help, thank you!!!

roceh, 2015-10-26:
One thing I recall when I built LLVM, it was building release version, but llilum was looking for the debug version (building llilum as release caused problems in itself), I had to change the cmake configure line in BuildLlvmWithVS.cmd from
```
cmake -G"%~1" -DCMAKE_INSTALL_PREFIX=Install ..\..
```
to
```
cmake -G"%~1" -DCMAKE_INSTALL_PREFIX=Install -DCMAKE_BUILD_TYPE=Debug ..\..
```
the --config Debug  on the build seemed to do nothing.

weixiongmei, 2015-10-27:
Hi roceh,

I still can't get it works, could you write a guide or record a video how to build this project? I followed the wiki, it's a bit confusing. really want to try it and port it to cortex-m0 microcontrollers. Thank you!

smaillet-ms, 2015-10-29:
What version of CMake are you using? It seems they changed some things in how the configurations are generated and run - the script was built and tested with 3.2.3. I'll need to update and test the script with a newer version.

weixiongmei, 2015-10-30:
it's working now, and tested it with couple samples, llium is still too "young" at the moment, can't use it for real project until the GC done. However, thanks you mr.smaillet-ms~~~

closed 2015-12-03 by ZachNL