NETMF/llilum #52: DIBuilder Error

weixiongmei, 2015-10-28:
I just got the LLVM successfully builded, but still having problems when build the CreateSDKDrop.bat & Zelig.sln, there are two errors always appear.

error: ‘class llvm::DIBuilder’ has no member named ‘createLocalVariable’
error: 'llvm:DIBuilder::createSubroutineType' can't accept 3 params.

Please advise how to fix it, thank you~~

roceh, 2015-10-28:
I used this version http://llvm.org/releases/3.7.0/llvm-3.7.0.src.tar.xz

smaillet-ms, 2015-10-28:
Interesting the mirror has a different API signature... I'll have to look into that and see. I used the GitHub Mirror while 3.7.0 was still under development, but the version we use now should be the official files from the llvm.org site directly.

smaillet-ms, 2015-10-28:
I just downloaded the source from LLVM Site  http://llvm.org/releases/3.7.0/llvm-3.7.0.src.tar.xz
and ran a full file by file diff against what we currently build with and it matches exactly, so we are building against the official release.

NOTE: The GitHub mirror will mirror the head of the development for whatever is the **next** release. (e.g. it does **not** contain the latest official released version but tracks active development of an unreleased version)

weixiongmei, 2015-10-28:
thanks mr.roceh & mr.smaillet-ms, i just rebuild the LLVM 3.7.0, and those two have gone now, but still can't get the SDK builded. A new error appears,

\r [E:\llilum-dev\Zelig\Zelig\Test\mbed\SimpleSDK\Native\Native.vcxproj] C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.Cpp.Platform.targets(57,5): error MSB4018: System.Form at Exception: Index(Starting From Zero) Must greater than or equal to Zero, and smaller than the param list.

Having a lot of troubles when building this project... :(

Skobovm, 2015-10-28:
Please unload that project until you have installed the SDK. In order to create the SDK, please follow the steps here: https://github.com/NETMF/llilum/wiki/SDK-User-Guide (with the Native project unloaded). OR leave the project unloaded, and follow the steps here, for using Llilum without the SDK and using makefiles instead: https://github.com/NETMF/llilum/wiki/demo

weixiongmei, 2015-10-28:
Hi Skobovm,

Still having the same problem even with the Native project unloaded in the Zelig.sln

Skobovm, 2015-10-28:
Where exactly are you in the setup process?

weixiongmei, 2015-10-28:
Hi Skobovm,

Just got it working, thanks for you help, i have to unload and remove the Native project from the Zelig.sln, just unload it is not enough, thanks again~~~

Skobovm, 2015-10-28:
No problem! Good luck

closed 2015-10-28 by Skobovm