NETMF/llilum #54: generic collections supported?

roceh, 2015-11-01:
Zelig seems to crash if you use generic collections (a dictionary in this case), seems to be at this spot (Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\LLVM\LLVMModuleManager_Types.cs:153)

The type in question is:
```
System.Collections.Generic.Dictionary`2<TKey,TValue>.Entry<TKey,TValue>&
```
In a previous commit you had a check there that if the underlying type was null it would return null. Putting that back in gets a bit further, but eventually I get an assert for this type: 
```
System.Collections.Generic.EqualityComparer`1<T>
```
assert being from LLVM\lib\IR\Type.cpp
```
"Expression isOpaque() && "Struct body already set!"
```
This class ( http://pastebin.com/gsptjk3D ) in the simple project generates these issues, add this the main in program.cs
```
 var font = Font.LoadFromFile("DEJAVU.FNT");
```

lt72, 2015-11-09:
Hi David,
collections are supported and so generics. We should have a [Dictionary](https://github.com/NETMF/llilum/blob/dev/Zelig/Zelig/RunTime/Framework/mscorlib/System/Collections/Generic/Dictionary.cs) that works. 
We want to look into this ASAP, but we are all a bit tied up at the moment. 
Are you blocked by this issue or can you get around it, e.g. by using List and Binary Search?

roceh, 2015-11-09:
Hi Lorenzo, Its not an urgent fix as I was just playing to see what was possible. The problem didn't always occur - sometimes dictionaries worked fine.

smaillet-ms, 2016-01-07:
Can you provide reproducible steps to get this? At the very least there's a bug in Llvm.NET here. Ideally Llvm.NET should be able to test for bad usage of the wrapped LLVM support and generate a proper .NET exception instead of letting things get all the way to the native code assert (Which would be disabled in a release build, usually ending in a crash, sometimes nowhere near where the problem really is)

roceh, 2016-01-08:
The first problem now does not occur, the 
```
"Expression isOpaque() && "Struct body already set!" 
```
still occurs. 

To replicate I just used the standard demo project (I used a K64F to test) that gets created when your create a llilum app with the VS extension , put the code that is in the pastebin link into a .cs file and then add the following

```
var font = Font.LoadFromFile("DEJAVU.FNT");
```

and include the using ....

I put the load statement here:

```
            int threadPinState = 1;
            int loopPinState = 1;

            var font = Font.LoadFromFile("DEJAVU.FNT");

            // Toggling a pin with a thread
            var ev = new AutoResetEvent(false);
```

and then compile, its gets to this point
```
1>      92.335 [   0.232]: Phase: GenerateImage [types: 383, fields: 703, methods: 1556]
1>       0.000163: START ##########################################################################
1>       0.000201: Round: 1
1>       0.000208: Emitting...
```

and then shows this dialog:  http://imgur.com/zbsNLlf

lt72, 2016-09-01:
I cannot repro this one. Please re-open in case.

closed 2016-09-01 by lt72