NETMF/llilum #56: GPIO incorrectly checks for failure while creating DefaultPinProvider

ColinA-MSFT, 2015-11-09:
Found while enabling redundant null dead-stripping.  In GpioController.cs in the following code, provider can never be null:
            IGpioPinProvider provider = new DefaultPinProvider(pinNumber);

            if(provider == null)
            {
                pin.Dispose();
                pin = null;
                status = GpioOpenStatus.PinUnavailable;
                return false;
            }

We need to try/catch the creation of DefaultPinProvider and/or change the way we create the provider.

lt72, 2015-11-10:
Mikhail, please take care of this issue with the same technique you are using for other similar types.

Skobovm, 2015-11-10:
Fixed this, and similar issues

closed 2015-11-10 by Skobovm

Skobovm, 2015-11-10:
Fixed in https://github.com/NETMF/llilum/commit/84306453b8e91067490da8d25a8a24e08b55c02e