NETMF/llilum #57: Creating a BSP fails due to make errors

cw2, 2015-11-10:
I am following the **Creating a Board Support Package** guide (\*) and I am stuck at point 11 of **Modifying the Makefile** section - it seems there are a few problems:
* `make STM32L152` ends with `error makefile:31: *** commands commence before first target.  Stop.`, because the spaces were (incorrectly?) replaced with tabs in a recent commit,
* after fixing that the error is `makefile:31: *** TARGET is undefined. Pass in TARGET for desired board.  Stop.`. I think the correct command is `make TARGET=STM32L152`,
* which fails with `make: *** No rule to make target .\Microsoft.Zelig.Test.mbed.Simple_opt.o, needed by STM32L152\mbed_simple.elf.  Stop.`, because Microsoft.Zelig.Test.mbed.Simple_opt.o is supposed to be created as a result of the Native project (in the next section), but the project cannot be loaded, as reported in the issue #39 (?)

(*) I have already fixed minor errors there

Skobovm, 2015-11-11:
Sorry about the confusion! There has been some churn in that area, recently, so the steps are already slightly outdated.  
  
For your first point, I will verify/fix any issues with the makefile. 
For the second point, you are correct. It needs to be invoked in the same way as it is in build.bat. I will update the instructions.
For your last point, you need to run setenv.cmd, found in the repo root, and that should resolve that issue. I will add that part to the steps.  
  
Thanks for trying out the steps, and let us know if you encounter any other issues!

Skobovm, 2015-11-11:
Correction: in the makefile, remove the line that adds the Microsoft.Zelig.Test.Simple_opt.o to the objects.

cw2, 2015-11-11:
Thanks for the hints.

While removing the makefile line that adds Microsoft.Zelig.Test.Simple_opt.o helps initially, it causes linker failure due to unresolved references (PendSV_Handler_Zelig, SVC_Handler_Zelig, HardFault_Handler_Zelig and a few more). So it has to be added back again, especially for build step 12 of **Testing the compilation** section.

With this workaround I've got mbed_simple.elf :)

One question, though: Are there any recommendations or plans how to deal with multiple platforms/boards in 'mbed/simple' project and such? I am going to add at least two more boards and merging everything into one makefile, program.cs with many #define-s is not going to be usable/manageable...

Skobovm, 2015-11-11:
You are correct, that makefile modification is temporary, and should only be used for step 11 of **modifying the makefile**.  Also, congratulations on the mbed_simple!
  
To answer your question: the makefile, simple program, and everything related is **only** there to help users get up and running quickly, or to test minor changes. For development, we recommend that everyone use the SDK. In the SDK, the "Native" project has a property section called "Llilum Configuration", where switching between boards is as easy as selecting one from a dropdown.  
  
Please see the [SDK User Guide](https://github.com/NETMF/llilum/wiki/SDK-User-Guide) for how to use the SDK, and continue with the guide you are on to finish up your BSP

cw2, 2015-11-11:
Thanks for the clarification, makes sense now :blush:

closed 2015-11-11 by cw2