NETMF/llilum #60: Unaligned memory access usage fault on STM32F1

cw2, 2015-11-11:
I have created a new BSP for [STM32F103 Nucleo](http://www.st.com/web/catalog/tools/FM116/SC959/SS1532/LN1847/PF259875?icmp=nucleo-ipf_pron_pr-nucleo_feb2014&sc=nucleoF103RB-pr) board and mbed_simple application (using the latest version of llilum) crashes during initialization due to unaligned memory access in MemoryFreeBlock.Allocate() method, line 133
```
ObjectHeader.CastAsObjectHeader( newAllocation ).InitializeAllocatedRawBytes( size );
 8004b00:	2800      	cmpne	r0, #0
 8004b02:	d00a      	beq.n	8004b1a <MemoryFreeBlock::Allocate#4903+0x7a>
 8004b04:	d009      	beq.n	8004b1a <MemoryFreeBlock::Allocate#4903+0x7a>
 8004b06:	210c      	movs	r1, #12
 8004b08:	2800      	cmp	r0, #0
 8004b0a:	6001      	str	r1, [r0, #0]
```
where it tries to store value of r1 = 0xC to address r0 = 0x200048BD, which is not aligned.

Is this related to issue #35?

jelin1, 2015-11-11:
Hmm. I'm not seeing this on other boards. At this point of the code, r0 should be newAllocation, which should always be 4-byte aligned, as should size (stored in r4). If r0 is not aligned, that likely means that a previous Allocate (or the InitializeFromRawMemory() if this was the first time this code is called) has left the free block in a bad state. The free block is represented as a byte array, and its length (num of elements) should always be 4-byte aligned.

cw2, 2015-11-11:
Ok, thank you for the information. I think that is the first Allocate call, so I'll have a look at memory state and what happens during InitializeFromRawMemory(). I understand it is almost impossible to help troubleshooting such issues when they cannot be reproduced (different hardware), so it is important to know it works fine on other boards...

closed 2015-11-11 by cw2

jelin1, 2015-11-11:
You'll want to make sure that in the InitilaizeFromRawMemory() call, both the baseAddress and sizeInBytes params should be 4-byte aligned.

cw2, 2015-11-12:
Well, it seems the problem was that I missed to update the value of Device.Instance.ManagedHeapSize, used in MemoryManager.InitializeMemoryManager - although I updated the makefile to pass correct heap size to native build, the MemoryManager used wrong value (too big).