NETMF/llilum #63: How much heap memory is required?

cw2, 2015-11-12:
Would it be possible to determine how much heap memory does the current runtime require? I am working on a BSP for STM32F103 that has 20 KB RAM (0x5000) and I am encountering some issues that could be explained by not having enough free memory (configured heap size to 0x4700, ~= 0x20005000 - __HeapLimit at 0x200008a0 in the .map file).

BTW, I must say the level of managed to native translation is impressive :bow:

lt72, 2015-11-13:
Hi cw2,
all the memory goes to the threads stack, and since GC is still disabled, you can never get it back.
We are working to limit the number of threads to the bare minimum right now, and we think we can get away with
- the main thread for the app => 256 bytes
- an idle thread with a tiny stack =>  2KB
- one thread to dispatch interrupts and timers => 2KB
- any thread you create => 2KB each

cw2, 2015-11-17:
Ok, thanks for the information.

The problems were caused by invalid heap memory configuration, I had to specify significantly smaller size based on return value of malloc() - while the size calculated from the map file was 0x4700, malloc() could allocate only 0x3FE7 (*). After setting heap size to 0x3E00 the board started working...

(*) The current implementation in `Memory.RequestMemoryPool()` decreases block size by 1/10 if it is too big, so this value is result of the second iteration (0x4700 - 0x4700/10), which succeeds.

Note regarding issue #60: I think it happens in case the malloc() fails to allocate heap memory block for requested size and when it uses decreased size (i.e. 0x3FE7) there are certain pointer operations later in the memory segment allocation methods which result in unaligned addresses - I forgot to keep my debugging notes, but if I remember it correctly, there are calculations relative to heap end and because that address was not aligned, the result was also not aligned.

closed 2015-12-02 by cw2