NETMF/llilum #65: K64F compilation fails with StackOverflowException

cw2, 2015-11-14:
The latest version of Zelig.Compiler fails with StackOverflowException when processing [unmodified] K64F config, at PrepareImplementationOfInternalMethods phase.

  Increasing stack size in the executable file header (`editbin /stack:0x800000`) helps to workaround this problem, but I think this is already being worked on (issue #59) (?)

  Meanwhile, are there any recommendations what can be changed to reduce the stack usage? E.g. remove reference assemblies, disable compilation phases and such?

Call stack (partial):
```
Microsoft.Zelig.CodeGeneration.CodeTransformation.dll!Microsoft.Zelig.CodeGeneration.IR.Transformations.ScanTypeSystem.Transform(ref Microsoft.Zelig.CodeGeneration.IR.BasicBlock bb) Line 1051	C# 
Microsoft.Zelig.CodeGeneration.IntermediateRepresentation.dll!Microsoft.Zelig.CodeGeneration.IR.UnconditionalControlOperator.ApplyTransformation(Microsoft.Zelig.CodeGeneration.IR.TransformationContextForIR context) Line 79	C# 	
Microsoft.Zelig.CodeGeneration.CodeTransformation.dll!Microsoft.Zelig.CodeGeneration.IR.Transformations.ScanTypeSystem.Transform(ref Microsoft.Zelig.CodeGeneration.IR.Operator op) Line 928	C# 	
Microsoft.Zelig.CodeGeneration.CodeTransformation.dll!Microsoft.Zelig.CodeGeneration.IR.Transformations.ScanTypeSystem.Transform(ref Microsoft.Zelig.CodeGeneration.IR.Operator[] opArray) Line 940	C# 	
Microsoft.Zelig.CodeGeneration.IntermediateRepresentation.dll!Microsoft.Zelig.CodeGeneration.IR.BasicBlock.ApplyTransformation(Microsoft.Zelig.CodeGeneration.IR.TransformationContextForIR context) Line 619	C# 	
Microsoft.Zelig.CodeGeneration.CodeTransformation.dll!Microsoft.Zelig.CodeGeneration.IR.Transformations.ScanTypeSystem.Transform(ref Microsoft.Zelig.CodeGeneration.IR.BasicBlock bb) Line 1051	C# 	
Microsoft.Zelig.CodeGeneration.IntermediateRepresentation.dll!Microsoft.Zelig.CodeGeneration.IR.UnconditionalControlOperator.ApplyTransformation(Microsoft.Zelig.CodeGeneration.IR.TransformationContextForIR context) Line 79	C# 	
Microsoft.Zelig.CodeGeneration.CodeTransformation.dll!Microsoft.Zelig.CodeGeneration.IR.Transformations.ScanTypeSystem.Transform(ref Microsoft.Zelig.CodeGeneration.IR.Operator op) Line 928	C#
```

Skobovm, 2015-11-14:
Hi cw2,  
  
This is a known issue, and we are investigating solutions. For the time being, there are no concrete workarounds that I am aware of.

cw2, 2015-11-14:
Thanks for the information.

closed 2015-11-14 by cw2