NETMF/llilum #67: Generic Tasks do not compile

Skobovm, 2015-11-17:
When compiling code with a generic Task, Llilum compiler fails on Armv7CompilationState::ConvertValueToStoreToTarget with "Invalid cast type"

Breaking code:
`Task<IBuffer> readTask = new Task<IBuffer>(() =>  `
`{  `
`                    return new ByteBuffer(1);  `
` });  `
`readTask.Start();`

ColinA-MSFT, 2015-11-17:
Additi

ColinA-MSFT, 2015-11-20:
I can't get a repro on this at present.  It may have been fixed by the phi work rearranging the graph.  I'll keep this open until I can ensure the root cause is addressed.

ColinA-MSFT, 2015-11-23:
As near as we can tell, this no longer repros.  I've tried several variations of the original code sample, both pre- and post-phi work, but I always get a clean compile.  More targeted attempts to get the system into a bad state have also proven unsuccessful.

closed 2015-11-23 by ColinA-MSFT