NETMF/llilum #71: Q: Adding support for ARMv6-M ISA ?

cw2, 2015-11-19:
I am curious what would be (the high-level) steps to add support for ARMv6-M ISA (translated by LLVM) and the necessary runtime?

smaillet-ms, 2015-12-02:
At present we don't isolate the instruction set target particularly cleanly. So it would be modifications to a number of places that assume the ISA. I'm cleaning some of that up as I go but there's no set time frame. I'm assuming you are after an M0 target - Which particular SOC are you after?

lt72, 2015-12-02:
the largest work items would be:
1) re-write the native portion of the CTX switch
2) create a Processor class with the right set of extensions into the system control registers for the M0 (not the same as M3/4) 
3) change interrupt enable/disable model (M0 has no BASEPRI, only PRIMASK) 

Likely a 3 to 4 weeks effort

cw2, 2015-12-02:
Thanks for the information. Yes, Cortex-M0, STM32F103 in particular (NUCLEO-F103RB).

If I understand it correctly, the tasks you mention are for the runtime support?

smaillet-ms, 2015-12-02:
Yes, Lorenzo's comments apply to the runtime (particularly the threading implementation) My comments were in regards to the compilation tools and code generation. 

Are you sure of those part numbers? The [STM32F103](http://www.st.com/web/catalog/mmc/FM141/SC1169/SS1031/LN1565) has a Cortex-M3 in it, which wouldn't require any changes to the code, though the RB variant might be a bit tight on memory right now as we haven't applied a lot of optimizations yet. (We are looking at various levels of optimization now so you should see continued improvement on that over time).

cw2, 2015-12-02:
Sorry, wrong board - the correct one is NUCLEO-F072RB, which has [STM32F072RB](http://www.st.com/web/catalog/mmc/FM141/SC1169/SS1574/LN1823/PF259605) Cortex-M0.

cw2, 2016-01-19:
Fixed by 29348c02d8ae1d577e900951bb43441f666cfe32

closed 2016-01-19 by cw2