NETMF/llilum #72: Q: Code style - spaces within argument list parentheses ?

cw2, 2015-11-19:
I apologize in advance, but I just have to ask: What's the deal with the spaces within argument list parentheses in the source code? Also, it's probably the most inconsistent style I've seen in a long while, with `some( spaces ) { nospaces(); weird( ( are( ) & these) || spaces); }`... Who writes like that?

Even the NETMF code looks... oh wait...

lt72, 2015-11-19:
same team, same style ;-)

closed 2015-11-20 by cw2

ColinA-MSFT, 2015-11-20:
Glad you asked!

The existing code has multiple sources, though the bulk of it is from the same team that built NetMF.  As Lorenzo notes, same team, same style.  Where possible, we've tried to stay consistent with the surrounding style within a file or project.

We do have plans to unify style across the code base to bring us closer to the VS defaults and Roslyn/CoreCLR.  We've been holding off on this until December, to minimize disruption while we get the major pieces in place.

Thank you for your patience and your feedback!