NETMF/llilum #73: Zelig is compiled successfull but no output files.

chutichnuoc, 2015-11-20:
I am following the link https://github.com/NETMF/llilum/wiki/demo, and get stuck at step Build Zelig image. It says "The build output directory will be located in <repo>\Zelig\LLVM2IR_results\mbed\simple\ " but after building without any error, there is no output files so I can not go to next step. I know I am missing something but can you please give me some hint?

Skobovm, 2015-11-20:
After building everything successfully, you need to set FrontEnd as the startup project, pass in the path to a configuration file in the properties, and run it.  
  
Unfortunately, some people have been encountering a StackOverflow issue from the Llilum compiler, so the current workaround is opening a VS Developer Cmd Prompt, and executing `editbin /stack:80000000 Microsoft.Zelig.Compiler.exe`, then running `<repo>\Zelig\ZeligBuild\Host\bin\Debug\Microsoft.Zelig.Compiler.exe -cfg <repo>\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\mbed_simple_LPC1768.FrontEndConfig`  
  
This should create the output files you are looking for.

chutichnuoc, 2015-11-20:
Thank you a lot, your solution worked for me.

closed 2015-12-03 by ZachNL