NETMF/llilum #75: Building SDK Error: Mbed Source and headers missing

chutichnuoc, 2015-11-23:
Hi,
We are on the way to learn more about LLilum.
- We are following SDK-User-Guide, at step Creating the SDK, we have problem "Mbed Source and headers missing", also mbed-rtos.lib not found (first picture)
- We did take a look at "Troubleshooting" at end of that page, it says "Be sure to select your board under Native -> Properties -> Llilum Config". But in my case it shows as the second picture, so we are not sure how to check it.
- If we want K64F as default device, should we add "#define K64F" in pch.h?
- Also in CreateSDKDrop.bat, 
we have: 
echo Copying board configurations
xcopy /i /e /q /y "%LLILUM_ROOT_DIR%\Zelig\Zelig\RunTime\DeviceModels\Boards" "%SDK_DIR%\SDKDrop\Boards"
But looks like "%LLILUM_ROOT_DIR%\Zelig\Zelig\RunTime\DeviceModels\Boards" folder does not exist? What should we replace to if we want K64F, please?
[image]
[image]
Thank again for your help!

Skobovm, 2015-11-23:
Hi,  
It appears that you are not synced up to latest. Please setup https://github.com/NETMF/llilum.git as the upstream repo, and do a `git pull upstream dev` in your local branch.  
  
More info is here: https://github.com/NETMF/llilum/wiki/Git-workflow  
  
Let us know if you have any more questions!

chutichnuoc, 2015-11-23:
Thank you too much, we passed that step by update to latest version.
Now we are installing the SDK and have another issue:
Error: MSBuild is not installed to where it should be. "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Application Type"

Thanks again for your help!

[image]

Skobovm, 2015-11-23:
Hmm, that's an interesting one. Can you verify that the path "C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Application Type" exists for you? If not, I believe the problem is that you do not have all of the prerequisites installed.