NETMF/llilum #77: SelfTest::SelfTest__Interlocked__CompareExchange_Object failed with RefCountGC turned on

jelin1, 2015-11-25:
I noticed that with refcountGC turn on, SelfTest::SelfTest__Interlocked__CompareExchange_Object will fail the following assert:

            Object o = new Object( );
            Object copy = o;
            Object sub = new Object( );
            BugCheck.Log( "Before: o = 0x%x, sub = 0x%x", (int)ObjectHeader.Unpack( o ).ToPointer( ), (int)ObjectHeader.Unpack( sub ).ToPointer( ) );
            Object oldo = Interlocked.CompareExchange( ref o, sub, copy );
            BugCheck.Log( "After: o = 0x%x, oldo = 0x%x", (int)ObjectHeader.Unpack( o ).ToPointer( ), (int)ObjectHeader.Unpack( oldo ).ToPointer( ) );

            BugCheck.Assert( o == sub && oldo == copy, BugCheck.StopCode.Impossible ); // <==

jelin1, 2015-11-25:
From the LL, it looks like it’s reusing the cached value of o, instead of reloading it after the call to CompareExchange.

ColinA-MSFT, 2015-11-25:
It looks like o gets split into two phi variables, one of which is on the stack and one of which isn't.  I suspect the right solution here is to force all the aliases for an addressed expression to reference the shared stack location.

commit 17a55b72 (2015-12-01): Remove caching of addressable values.

Fixes #77
closed 2015-12-01 by ColinA-MSFT