NETMF/llilum #84: Llilum Debug: Unable to start because can not file specified.

chutichnuoc, 2015-11-30:
Hi, after installed SDK, I created my first project, compiled and deployed to device successful
[image]
. 
The next thing we have to learn is debugging by VS2015. 
I am following the link: https://github.com/NETMF/llilum/wiki/demo, at end of the page, and it gives me error: "Unable to start debugging. The system cannot find the file specified" as picture below. I checked and those files exist and path is correct.
There is aslo one thing looks like not clear too much. It is the statement: "To debug in Visual Studio 2015:  Launch VS from the command line so that it picks up your local environment variables."
Do you mean run VS from the command line after called C:\src\llilum\setenv.cmd? 
Thank you to your help!

chutichnuoc, 2015-11-30:
Looks like I got it from the link: https://github.com/NETMF/llilum/wiki/SDK-User-Guide
Thank you!

Skobovm, 2015-11-30:
Hi, you do not need to follow those steps anymore. Pressing F5, or clicking on Debug should work.  
  
One note: if you encounter the issue where your breakpoints are not being hit, type in `Debug.MIDebugExec monitor reset halt` into the Command Window, and then hit F5 - this should reset the board, and load all of the breakpoints.  
  
Let us know if you have any other issues!

closed 2015-12-03 by ZachNL