NETMF/llilum #85: Revisit RemoveRedundantBoundChecks optimization.

ColinA-MSFT, 2015-12-01:
The RemoveRedundantBoundChecks optimization doesn't appear to function on LLVM, as it searches for lower level operators that are only generated by the legacy system.  While we don't appear to generate unnecessary IndexOutOfRange checks, we may gain some small performance benefit from fixing this transform and rolling it into the more general RemoveRedundantChecks in the same file.  Otherwise, it should be removed.

lt72, 2015-12-01:
I would actually prefer to see this done before December 18th 
It is an important step in making sure that we have a full e2e working model and bounds checks in low level part of the system could be very damaging.
Also, we should look into making sure that removal of redundant NULL checks works well!

ColinA-MSFT, 2015-12-02:
Agreed.  At present, we only have a handful of bounds checks in the sample apps, but if we're adding them to critical paths that does push up the priority.

Redundant null checks are working very well.  In fact, the way these are implemented isn't limited to pointers, so we remove a lot of always-false branches for bools and other scalar constants, including a few critical paths (like allocation).

lt72, 2015-12-02:
Excellent, so let's try and do this by 12/18. Since the low level operators are not available, we try and re-work the logic around the higher level operators. Should be n interesting work item.