NETMF/llilum #86: Can not debug after download latest version.

chutichnuoc, 2015-12-01:
Hi,
It worked fine before but now when I try it gives me an error (as below) when the board is running.
I updated latest version, reverted all what I changed, but it still throw that error.
If you know what the problem is, please give us a help.
Thank you too much!
[image]

Skobovm, 2015-12-01:
Try to click Break, then in the Command Window type in `Debug.MIDebugExec monitor reset halt `, and then hit F5. This should restart the board, then another one of these windows will show up. Click continue in the pop-up, and debugging should work.  
  
Also, take a look at the callstack - you might be getting an actual exception somewhere in your code. If this issue persists, please let us know, and we will investigate further.

chutichnuoc, 2015-12-02:
Hi, after few tried, we think the problem is in TryOpenPin(pinNumber, sharingMode, out pin, out status); The kind of exception is OutOfRange. We tried Pin.D8 (on K64F this is the first pin PTA0) so it should not be out of range. Looks like there is a racing between creating heap and objects, because there is no changed and source code is up to date. If you have any suggestion, please share it to us
Thank you too much!
[image]

ZachNL, 2015-12-02:
I can reproduce this problem as well.  I will look into it and get you a fix as soon as possible.

ZachNL, 2015-12-02:
I saw the same problem you are seeing, but this was because I forgot to swap the defines at the top program.cs.  Did you make sure to comment out #define LPC1768 and uncomment #define K64F?  Unfortunately this is necessary because the Llilum properties for selecting the board are in the Native project and does not affect the Managed project.  Therefore it has to be changed manually.

chutichnuoc, 2015-12-02:
I think I already did it. I also switch to K64F in Native ->Llilum Configuaration.
Not sure what I am missing now.
[image]

ZachNL, 2015-12-02:
If you haven't already, you could try to rebuild just the Native project to make sure Llilum compiler is re-run (right click on the Native project and select "rebuild").  Also can you see what pin number it is failing on?  If it is not 4118 or 4117, then you are likely using the LPC1768 pins.

chutichnuoc, 2015-12-02:
Hi, thanks again for trying to support us.
I checked again and it is 4117 
My another question is,  is variables watching available now? As you can see in the picture, looks like this feature is not supported or am I doing something wrong?
Do we need to config any special thing before compile SDK? I remember we don't need but hopefully I am wrong.
[image]

ZachNL, 2015-12-02:
OK, I took a closer look at your stack trace and see that you are getting a failure during the object synchronization for try/finally.  My apologies, my first guess was completely wrong but happened to be at the same managed code instruction.  I am not sure what to suggest at the moment other than uninstalling the SDK, closing all VS instances, and rebuild and re-install the SDK.  I can not reproduce the problem.

I also asked the debugger expert about the variables/watch window and he said they should be working (although I too don't see anything).  I will have the expert investigate.

smaillet-ms, 2015-12-14:
Out of curiosity, what version of VS do you have installed (in particular do you have Update 1?). I'm seeing a similar situation but only on a system building everything with VS Update 1 installed - Still investigating this but wanted to see if this was another "data point". 

As to the watch and locals window, there was a change to the code generation that optimized the way we handle locals (and in particular temporary values) while this saved a good bit of code size it also had the unfortunate side effect of losing the debug tagging for the local variables and arguments. That has to be handled a bit differently with the new code generation pattern, I'm working on that.

chutichnuoc, 2015-12-14:
Yes, it worked fine before I updated to "Update 1". I also thought about this, but when I reinstalled VS, the "Update 1" is default added automatically, and noway to uncheck it.

smaillet-ms, 2015-12-14:
OK, thanks for the confirmation. Unfortunately once you've got update one there's no way I know of to uninstall it and there's no option I am aware of to install on a new machine without it. We're actively investigating a resolution to why the Llilum compilation behaves differently with Update1.

smaillet-ms, 2015-12-16:
With commit 7f8a3b25704f401da2e066ab0dafd363c2c25175 this should now be fixed. Thanks for helping confirm the issue. It was a "fun" one to track down, but happily a simple one to fix.

closed 2015-12-16 by smaillet-ms

chutichnuoc, 2016-01-04:
I just downloaded latest version today and I think the bug is still there. These were our steps:
- Download latest version
- Compile SDK
- Uninstall old SDK, install new one
- Create new project, select Llilum project template
- In Managed Project, disable LPC1768, enable K64F
- In Native, at Llilum Configuration, switch from LPC1768 to K64F
- Rebuild all - solution
- Run and debug on real device K64F. 
Nothing change, the bug is still there. Can you please double check again, please?

smaillet-ms, 2016-01-07:
When you say "download latest version" what exactly do you mean. Can you provide detailed steps used? If you already had a fork or cloned repo it's possible some special steps are needed as there was a case of the repository getting a bit messed up and the fix required some re-writing of the history. If you just created a completely fresh, clean clone in a new directory you won't be affected by the history re-write. I've just re-verified this using K64F and the sample blinky application is working.

chutichnuoc, 2016-01-07:
The problem was "VS update 1" issue. 
Then probably I didn't clean up my project enough.
Everything works fine now. 
Thanks to LLilum team.

smaillet-ms, 2016-01-07:
Phew - I was a little worried there was more to this than the issue I found and fixed.