NETMF/llilum #87: Boolean flag not needed on:  ```public object InitializeObject(UIntPtr memory, TS.VTable vTable, bool referenceCounting)```

lt72, 2015-12-02:
No need for bool flag, just need to pass in the flags as a parameter from 
public override Object AllocateObject( TS.VTable vTable ) and 
public override Object AllocateReferenceCountingObject( TS.VTable vTable )

this will eliminate one costly branch on each object allocation

        public object InitializeObject(UIntPtr memory,
                                        TS.VTable vTable,
                                        bool referenceCounting)
        {
            ObjectHeader oh = ObjectHeader.CastAsObjectHeader(memory);

            oh.VirtualTable = vTable;

            if (referenceCounting)
            {
                oh.MultiUseWord = (int)((1 << ObjectHeader.ReferenceCountShift) | (int)ObjectHeader.GarbageCollectorFlags.NormalObject | (int)ObjectHeader.GarbageCollectorFlags.Unmarked);
#if REFCOUNT_STAT
                ObjectHeader.s_RefCountedObjectsAllocated++;
#endif
#if DEBUG_REFCOUNT
                BugCheck.Log( "InitRC (0x%x), new count = 1 +", (int)oh.ToPointer( ) );
#endif
            }
            else
            {
                oh.MultiUseWord = (int)(ObjectHeader.GarbageCollectorFlags.NormalObject | ObjectHeader.GarbageCollectorFlags.Unmarked);
            }

            return oh.Pack();
        }

jelin1, 2015-12-02:
Since InitializeObject is inlined, and the caller (AllocateObject / AllocateReferenceCountingObject)always uses a constant true/false, there's actually no branch here.

closed 2015-12-02 by jelin1

lt72, 2015-12-02:
Opts take care of it, so that is good