NETMF/llilum #91: CodeGen issue with using scope

ZachNL, 2015-12-07:
File: EventWaitHandleImpl.cs 
Function: public override bool Acquire( SchedulerTime timeout )

One of the two "using(SmartHandles.InterruptState.Disable())" blocks is exited without running dispose.  This code is executes during context switching and will be hit if the thread is waiting on an event is in the process of going to sleep when the event is set (on a separate ISR procedure).  This leaves the interrupt in the disabled state when exiting this function which should never happen.

The problem appears to be related to the return statement inside the using blocks.  If I set a Boolean instead of returning and then return after the using block, everything seems to work properly.

Original Code:
```
                    using(SmartHandles.InterruptState.Disable())
                    {
                        if(m_owner.m_state)
                        {
                            if(m_owner.m_mode == System.Threading.EventResetMode.AutoReset)
                            {
                                m_owner.m_state = false;
                            }

                            return true;
                        }
```
Code that works properly:
```
                    using(SmartHandles.InterruptState.Disable())
                    {
                        if(m_owner.m_state)
                        {
                            if(m_owner.m_mode == System.Threading.EventResetMode.AutoReset)
                            {
                                m_owner.m_state = false;
                            }

                            fExit = true;
                        }
                    }

                    if( fExit )
                    {
                        return true;
                    }
```

ZachNL, 2015-12-07:
Disasm file for the offending function: EventWaitableObject::Acquire

[bug.disasm.txt](https://github.com/NETMF/llilum/files/54598/bug.disasm.txt)

ColinA-MSFT, 2015-12-07:
It looks like we're doing the right thing.  If we enter the "if(m_owner.m_state)" block, we'll always jump to an inlined InterruptState::Dispose call before returning:
```
"BasicBlock(102)":                                ; preds = %"BasicBlock(101)"
  %"Local_43(Microsoft.Zelig.Runtime.SmartHandles.InterruptState).m_basepri" = getelementptr %Microsoft.Zelig.Runtime.SmartHandles.InterruptState, %Microsoft.Zelig.Runtime.SmartHandles.InterruptState* %"Local_43(Microsoft.Zelig.Runtime.SmartHandles.InterruptState)", i32 0, i32 0, !dbg !8953
  %24 = load i32, i32* %"Local_43(Microsoft.Zelig.Runtime.SmartHandles.InterruptState).m_basepri", !dbg !8953
  %25 = call i32 @CMSIS_STUB_SCB__set_BASEPRI(i32 %24), !dbg !8953
  br label %"BasicBlock(24)", !dbg !8953

"BasicBlock(24)":                                 ; preds = %"BasicBlock(27)", %"BasicBlock(23)", %"BasicBlock(55)", %"BasicBlock(53)", %"BasicBlock(102)"
  %26 = phi i8 [ %55, %"BasicBlock(23)" ], [ %55, %"BasicBlock(27)" ], [ 1, %"BasicBlock(53)" ], [ 1, %"BasicBlock(55)" ], [ 1, %"BasicBlock(102)" ]
  br label %"ExitBasicBlock(25)", !dbg !8957

"ExitBasicBlock(25)":                             ; preds = %"BasicBlock(24)"
  ret i8 %26
```

ZachNL, 2015-12-15:
I pasted the incorrect block that was causing the problem.   The code has been updated in this bug report (above).

lt72, 2016-01-12:
still happening, with exactly the same description as Zach's

ColinA-MSFT, 2016-01-13:
TransformFinallyBlocksIntoTryBlocks mishandles leave operators.  In cases where the operator jumps out of an outer nested exception context when both the inner and outer contexts have finally handlers, we skip the inner one.

commit 9573e59c (2015-12-31): Reduce complexity of SSA transform output:
- Don't split exception handling blocks for LLVM.
- Convert all phi node variables to addresses, removing the need for phi nodes entirely.

Fixes #91, #107
commit 54a4ce60 (2016-01-13): Ensure leave operators unwind all surrounding finally blocks.

Fixes #91
closed 2016-01-13 by ColinA-MSFT