NETMF/llilum #93: Cannot implicitly call reference counting methods, any GC call need to go through manager

lt72, 2015-12-12:
While setting up the infrastructure for Boehm collector, I found this:

Break () at mbed_debug.cpp:33
Breakpoint (n=<optimized out>) at mbed_debug.cpp:50
BugCheck::Raise#5142 ()
TypeSystemManager::Throw#5456 () at TypeSystemManager.cs:368
ProcessorARMv7M::RaiseSupervisorCall#3488 () at ThreadManager.cs:239
BoehmCollector::Collect#4614 () at BoehmCollector.cs:527
DefaultTypeSystemManager::AllocateInner#5472 () at DefaultTypeSystemManager.cs:164
DefaultTypeSystemManager::AllocateReferenceCountingArray#5466 () at DefaultTypeSystemManager.cs:77
Thread::.ctor#4261 () at ThreadImpl.cs:64
InterruptController::Initialize#18058 () at InterruptController.cs:170
Peripherals::Activate#17945 () at Peripherals.cs:49
Peripherals::Activate#17696 () at Peripherals.cs:29
Bootstrap::HardwareActivation#3951 () at Bootstrap.cs:160
Bootstrap::Initialization#3945 () at Bootstrap.cs:93

and then this:

 [CompilationSteps.PhaseFilter( typeof(Phases.HighLevelTransformations) )]
        [CompilationSteps.OperatorHandler( typeof(ArrayAllocationOperator) )]
        private static void Handle_ArrayAllocationOperator( PhaseExecution.NotificationContext nc )
        {
            ArrayAllocationOperator         op                  = (ArrayAllocationOperator)nc.CurrentOperator;
            TypeSystemForCodeTransformation ts                  = nc.TypeSystem;
            TemporaryVariableExpression     typeSystemManagerEx = FetchTypeSystemManager( nc, op );
            MethodRepresentation            mdAllocateArray     = ts.WellKnownMethods.TypeSystemManager_AllocateArray;
            CallOperator                    call;
            Expression[]                    rhs;
            Expression                      exLen = op.FirstArgument;

            if(ts.IsReferenceCountingType( ts.WellKnownTypes.System_Array ) && !ts.ShouldExcludeMethodFromReferenceCounting( nc.CurrentMethod ))
            {
                mdAllocateArray = ts.WellKnownMethods.TypeSystemManager_AllocateReferenceCountingArray;
            }

please note that I am setting up a GC manager that is not reference counting, but we have code that does not depend or gets injected without going through the GC manager.

We need to make the ref counting GC manager a standard reference counting manager, so that changing GC manager does not leave behind any traces.

jelin1, 2015-12-12:
I'm trying to understand what's going on here. Do you have the InitializeReferenceCountingGarbageCollection phase enabled? if not, then ts.IsReferenceCountingType() should always return false, so no reference counting code will get injected.
If you do have it enabled, then it's technically by design for now.

That said, we need to solve the GC configuration issue, i.e. we need a way that will allow us to swap out GC policies with configuration. The current way with the GC managers is that it's tied in to the platform, so to change it, we need to modify the platform code to have a non-abstract class of the collector we want.

Breakpoint (n=) at mbed_debug.cpp:50

lt72, 2015-12-14:
I fixed the issue for now using TypeSystem.IsReferenceCountingEnabled, which may not be elegant but works.
I agree we need to fix it by using some compile-time strategy that will allow us to swap out GC policies with configuration. 
Let's think about that together, then I'll take care of it while you implement the Boehm collector.

closed 2016-01-22 by lt72