NETMF/llilum #95: low power

josesimoes, 2015-12-14:
Hi,

As wasn't sure of this by looking at the sources I thought I should ask here: is llilum going to support/handle properly low (really, really low!) power modes and sleep/wake-up cycles, etc? 

Thanks,

lt72, 2015-12-14:
We do support WFI/E at the moment, there is not other standard power cycle as of now in ARM Cortex.
We do not currently have an API to wrap any non-standard implementation either, but we could write one following NETMF style. Did you have any specifics in mind?

josesimoes, 2015-12-14:
Hey Lorenzo.

I really think that providing an API (even minimal) would be great. NETMF style maybe a good starting point but it doesn't provide all the required features for a usable sleep/wake-up scenario. (at least from my perspective...) Let me detail this better.
Putting the mcu to sleep should be quite straightforward. Having support for different power levels would be nice. 
Setting the wake-up event to be either an I/O transition and/or RTC alarm should suffice. Those are supported by the vast majority of mcus out there and should cover the vast majority of scenarios. 

Looking into the current NETMF we can see that it allows you to call CPU_Sleep and it's up the hardware port to implement it the correct way. For the current STM32F4 what it's missing is a true RTC support (I've just sent a PR to cover that). The part that is missing is how to set the wake-up source. My current approach for this is a Interop lib that sets the STM32F4 RTC alarm with a DateTime. At the set time the RTC resets the CPU and NETMF boots. 
This last part is missing from the NETMF API. The way I see it any platform that targets IoT, battery powered devices, wearables and such has to be "power aware" and to provide the developer means to fully manage that. 

Let me know if this makes sense to you. If not, I can try to explain it better. :smile:

lt72, 2015-12-14:
Did you take a look at the NETMF class below:
```
class PowerState
{

        public static void Sleep(SleepLevel level, HardwareEvent wakeUpEvents);
}
...
public enum HardwareEvent
    {
        SerialIn = 0x00000001,
        SerialOut = 0x00000002,
        USBIn = 0x00000004,
        USBOut = 0x00000008,
        SystemTimer = 0x00000010,
        Timer1 = 0x00000020,
        Timer2 = 0x00000040,
        Socket = 0x00004000,
        Spi = 0x00008000,
        Charger = 0x00010000,
        OEMReserved1 = 0x00020000,
        OEMReserved2 = 0x00040000,
        FileSystemIO = 0x00080000,
        GeneralPurpose = 0x08000000, // GPIO, Touch, Gesture, Storage, Network, etc
        I2C = 0x10000000,
    }
```

that clearly list the wake-up source.  What am I missing?

josesimoes, 2015-12-14:
What is missing is the implementation...
As far as I know there is no port/solution that implements those. Moreover I'm not aware of any "boot" options in NETMF other than cold boot. When you are waking up from low power modes (except deep sleep and such) ideally you don't have to restart everything from zero. There is no support for knowing the boot reason too.
Also there is no core support for automatically manage the power mode when the CPU is idling, etc.

lt72, 2015-12-15:
On the old codeplex site you will find the complete implementation for the API above, for most HW ports supported at that time: http://netmf.codeplex.com/ 

I suggest to look at MC9328MXL port to start. That was the i.MXS ARM9 from Freescale.

josesimoes, 2015-12-21:
@lt72 I've browsed carefully the codeplex repository as you suggested. My findings are:
1. I've found several MCU specific implementations for the most common sleep modes ("sleep", "deep sleep" and "shutdown") I'm referring to them using "" because each vendor uses different names for equivalent functionality.
2. For most of them the "sleep" and "deep sleep" are implemented allowing wake-up from external I/O interrupt, nothing else.
3. The .NETMF pretty much API exposes all the available options to setup the hardware event that should wake the cpu through the enum HardwareEvent.
4. It's up to each hardware port to actually make that happen. (the vast majority of them are not implemented on any hardware port).

Getting back to the original question and your follow up:
a. Yes, I think there should be an API "following NETMF style" as you mentioned.
b. Specifically about waking up from RTC I'm sure if HardwareEvent.SystemTimer is the one.
c. The API should provide a method to setup the "alarm time" that will wake up the cpu. As I've said before almost all mcus support this through their RTC module. In the end it will be up the port specific code to actually implement this, if it's of interest.
d. Almost all mcus have a register that provides the boot reason (power, reset, rtc, interrupt, hardware fault, etc). I think the API should expose this. Again, it will be up to each specific port to actually  implement it.

josesimoes, 2016-02-12:
@lt72 have you had the opportunity to give this some though? The same for issue #102 (which is kind of related). I've peeked into the code and it looks the hardware RTC isn't being used at all, but I might be wrong about this...