NETMF/llilum #103: Boxed values returned from methods cannot be unboxed.

lt72, 2015-12-22:
Suppose you have a method like

object ReturnInteger()
{
  int i = 42;
  object o = i;
  return o;
}

then this code woudl throw an InvalidCast exception:

int ii= (int)ReturnInteger();

ColinA-MSFT, 2015-12-28:
We appear to be translating boxing of value types incorrectly:
```
  %0 = call %System.Object* @"Microsoft.Zelig.Runtime.DefaultTypeSystemManager::AllocateObject#5573"(%Microsoft.Zelig.Runtime.DefaultTypeSystemManager* getelementptr inbounds (%Microsoft.CortexM3OnMBED.TypeSystemManager, %Microsoft.CortexM3OnMBED.TypeSystemManager* @Microsoft.CortexM3OnMBED.TypeSystemManager_10, i32 0, i32 0, i32 0), %Microsoft.Zelig.Runtime.TypeSystem.VTable* @System.Int32.VTable_127), !dbg !6797
  %1 = bitcast %System.Object* %0 to %System.Object**
  store %System.Object* inttoptr (i32 49 to %System.Object*), %System.Object** %1
```

This translates roughly to:
```
  Object* obj = new <boxed int>;
  *((Object**)obj) = (Object*)49;
```

ColinA-MSFT, 2015-12-30:
Fixed with the above linked change.  This change doesn't address all box/unbox issues, but a followup change will fix the remaining virtual call type mismatch.

closed 2015-12-30 by ColinA-MSFT