NETMF/llilum #105: Unable to Build 2061 errors on LibLLVM

danwalmsley, 2015-12-22:
[image]

Is there a missing step in the instructions?

I have built LLVM and not opened the zelig solution and right clicked solution, build solution..

Skobovm, 2015-12-24:
Can you verify that your registry has an entry in HKCU\Software\LLVM\3.7.0 called "SrcRoot"?

danwalmsley, 2015-12-24:
Yes its there..

danwalmsley, 2015-12-24:
[image]
points to this...

Skobovm, 2015-12-24:
This may seem silly, but can you clean the solution, and hit "CTRL + SHIFT + B" a few times to see if the error count goes down? Visual studio likes to act up at times. You can also try to build projects individually until one fails, and attempt to determine the cause.
  
If not, try cloning the repo to a different location and trying the steps again, without compiling LLVM (compiling LLVM is the long step, everything else should be quick). 
  
Unfortunately, I have never encountered the issue you are seeing, so I can't be of much help.

smaillet-ms, 2015-12-31:
Did you run CMAKe manually or via the LLVM build script provided? It looks like the source headers and compiled binaries aren't sharing the same set of preprocessor conditionals so you get those mismatches.

danwalmsley, 2016-01-05:
Hi, I'm just trying again... will see if the issue persists for me.

danwalmsley, 2016-01-05:
right I started from scratch and now it works, thanks for help guys!

closed 2016-01-05 by danwalmsley