NETMF/llilum #117: Unable to start debugging Unexpected GDB output -target-select remote :3333

danwalmsley, 2016-01-05:
[image]

I get this when trying to debug STM32 with ST Link.

Skobovm, 2016-01-05:
Hi Dan,  
This is an issue from OpenOCD not starting correctly. Please go to the Native project properties, Debugging tab, and check the following:  
  
* "Llilum Open Ocd" points to the correct location, and that OpenOCD was copied to that location
* ***(Most likely culprit)*** "Llilum Open Ocd Args" should look like  `-f $(LlilumSDK)tools\openocd\scripts\interface\$(LlilumOpenOcdInterfaceCfg) -f $(LlilumSDK)tools\openocd\scripts\board\$(LlilumOpenOcdBoardCfg)` If you changed this, make sure you have the -f params in the right places
* "Llilum GDB Path" is correct and the file exists  
  
Lastly, just to be safe, right click on the Native project, Open Project Directory, and open the tmp_debug.bat file in there. If any of the paths look incorrect, or the files don't exist, make the appropriate changes in the debugging properties, delete the tmp_debug.bat file, and try to debug again.

danwalmsley, 2016-01-05:
Hi, thanks I manually changes the Open OCD args to -f -f $(LlilumSDK)tools\openocd\scripts\interface\stlink-v2.cfg -f $(LlilumSDK) tools\openocd\scripts\target\stm32f4_stlink.cfg and had success, I couldn't see where those variables get defined.

That works now.

closed 2016-01-05 by danwalmsley