NETMF/llilum #118: Command Line Arguments for Compiler

danwalmsley, 2016-01-05:
I am interested in compiling a very minimal application with no board support.

Literally just the Zelig support stuff and a int Main() function in managed code.
 I would like to invoke the compiler manually, is there any information on what the command line options are?

Skobovm, 2016-01-05:
Are you referring to command line options for the Llilum compiler, or for the final compilation stage with GCC? And for no board support, do you just mean not use peripherals, do some sort of platform independent compilation? I don't think the latter is possible.  
  
If you are referring to the Llilum compilation options, please have a look at any of the *.FrontEndConfig files in the repo. Those are the command line arguments that you can pass in. You can try turning things on and off in there, but you will always need to have HostAssemblyDir, DeviceAssemblyDir, and CompilationSetup. For more info, you can look through Bench.cs in the FrontEnd project   
  
If you want to know more about the LLVM or GCC command line options, go to `<repo root>\Zelig\LLVM2IR_results\mbed\simple` and look at the build.bat file and makefile in there. The makefile is mostly a copy of the one from mBed.

danwalmsley, 2016-01-05:
Im really after using C# for as close to bare metal as possible, i.e. No kernel, nothing if that is possible, or absolutely minimal kernal but no mbed drivers.

Im talking about the zelig compiler, im not sure what its inputs are, does it take c# code and give me llvm object code out? What would be an example to compile an static int Main() program.cs that does nothing as a template, from command line, i.e 

My inputs are just startup.s for the device chip and the program.cs + any other minimal required files to get it to the Main function.

Skobovm, 2016-01-05:
Got it. What are your main concerns? Speed, flash, ram? The closest to bare metal you are going to get at this time, is just by not using any peripherals. The way that this system works is, it takes your C# code, and figures out what is and isn't being used. Once it does that, it pulls out all of the unused functions, classes, variables, etc. So, by not using, say Windows.Devices.Gpio, you will not pull in any of the GPIO code, and it will not get linked in.  
  
At this point in time, though, we are working on the Llilum runtime as a whole, so there is no good way to get rid of the Kernel code, unless the Llilum compiler itself removes it for you. Another thing to consider: if you are aiming for a smaller footprint, and still want peripherals, use Microsoft.Llilum.Devices rather than Windows.Devices, and don't use GarbageCollection if you don't need to. 

Something else you can look into is using interop functions (there are examples throughout the code base). They allow you to call into C from C#, and vice versa. This way you can call into your existing libs from C#, or do anything you don't want to deal with in C, in C#.

Lastly, if you really REALLY don't want to use mBed code, you can write your own OS Abstraction. Just create a copy of the MbedOSAbstraction, call it something else, stub out all of the functions, and only implement what you need to. Note: you will need to implement pretty much everything, except for the peripherals, so this approach might not be worth your time.

danwalmsley, 2016-01-05:
@Skobovm many thanks for taking the time to explain all this, very helpful that is exactly what I wanted to know.

   It has been a dream of over 10 years to be able to take C# and compile to ARM thumb.

Ok the solution you give to abstract write an OSAbstraction seems satisfactory.

The reason I am also interested in operating the compiler from command line is to possibly integrate LLilum into my cross platform IDE project, https://github.com/VitalElement/AvalonStudio, it is still a long way from complete but can already compile and debug C/C++ projects for these kinds of targets and to be able to support LLilum early on would be nice.

I tried passing in a CS file for to the compiler, but it gives me errors about argument null, it seems it expects 2 paths?

Is there a way to view the command line Visual Studio passed in when it compiles?

danwalmsley, 2016-01-05:
Oh I also noticed when using zelig from VS the process of compiling was very slow, like 2 minutes to compile the basic pin toggling application, mostly in the part where its stripping out stuff and converting to LLVM code, is that you experience or could it be a problem with my machine?

Perhaps its because I'm using the Debug build of the Zelig compiler?

Skobovm, 2016-01-05:
Glad I could help! For your compiler question: the Zelig compiler takes in a prebuilt executable. So if you want to do it from the command line, you will first have to compile the C# using MSBuild (have a look at `repo\Scripts\CreateSDKDrop.bat` for how to do that). That script compiles the entire solution (which makes things easy), so if you want to integrate MSBuild into your IDE, you will have to do some research on how to do so properly.
  
Then, the Zelig compiler takes in `-cfg repo\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\mbed_simple_XXX.FrontEndConfig` The .FrontEndConfig file actually says where to find the executable, where the output should go, etc. I believe you can also pass in most of the params in the FrontEndConfig files as command line arguments.
  
For the VS being slow question: unfortunately, I experience the exact same thing. It is on our radar of issues to address, however core functionality takes priority over convenience.

danwalmsley, 2016-01-05:
@Skobovm many thanks you guys are doing excellent work. Ok I will take on board all your advice and try out some of this. If I manage to get an integration to work I will let you know (could take a month or 2).

Is it / could it in future be possible to take in a .cs file, compile to MSIL with csc, then use zelig to convert the MSIL to a native .o to be linked manually?

danwalmsley, 2016-01-05:
I had a go at compiling manually a file with no references and just a main file...

[image]

I removed the references from the fontendconfig files also... so not sure where the System.MetaDataVersion reference comes from?

danwalmsley, 2016-01-05:
ok I got by that one by setting 
-HostAssemblyDir   ..\Debug
-DeviceAssemblyDir ..\..\..\Target\bin\Debug

danwalmsley, 2016-01-06:
ok I managed to get some success...

I created a class library, set output type to Console Application, then implemented static int main method.

I compiled using zelig, now I get it to build successfully, I'm guessing one of the output files I can get LLVM to convert to object code then link with GCC?

Any clues on which of the output files are relevant and what I need to do with them?

danwalmsley, 2016-01-06:
ok I found it... I will post incase its useful for others.

next step is:
llc -O0 -code-model=small -data-sections -relocation-model=pic -march=thumb -mcpu=cortex-m4 -filetype=obj -mtriple=Thumb-NoSubArch-UnknownVendor-UnknownOS-GNUEABI-ELF -o=program.o program.bc

that generates program.o...

now to link and debug... obviously I will need some startup code.

danwalmsley, 2016-01-06:
@Skobovm ok so now I'm almost there... except at link stage I get 

```
G:\vebuildLLilum\LLilum/mbed_core.cpp:455: undefined reference to `SVC_Handler_Zelig'
```

I passed in "-Reference Microsoft.Zelig.Runtime "

as an argument which I believe is the assembly that exports this...

and I ran objdump on the object file produced, which shows:
```
00000500 l     F .text.llilum	0000007c Microsoft.Zelig.Runtime.TargetPlatform.ARMv7.ProcessorARMv7M.Context::SVC_Handler_Zelig#6570

```

So it seems strange the linker cant find this, iv checked link order. Any ideas?

danwalmsley, 2016-01-06:
ok solved that one.... it required __FPU_USED to be set to a value other than 0!

Skobovm, 2016-01-06:
Sounds like you got it :)

danwalmsley, 2016-01-06:
@Skobovm 

[image]

Ok finally got it to link... file size is quite large? 

Now to see if it runs.

Skobovm, 2016-01-06:
Yep, it's going to be large for a little while. That image contains a memory manager, thread manager, a running idle thread, and some other stuff. As I said previously, our (current) goal is to create a fully functional runtime first, and then we will do work to optimize certain aspects of it

danwalmsley, 2016-01-06:
Does C# absolutely require a kernel and threads, etc, or could it in theory work without that? I understand why you absolutely need some kind of garbage collections, but you would only need a kernel once threads have been used?

danwalmsley, 2016-01-06:
I'm only thinking if there would be a way for me to cut that part out for experimentations.

danwalmsley, 2016-01-06:
I have it compiling using LLilum in my ide project now :D 

[image]

will get the debugger up and running now.

I would close the issue, but where can I contact you guys for help with some of this? i've opened an issue to open a gitter room which is handy for this kind of communication, and other .net repos have this.

Skobovm, 2016-01-06:
C# does not require threads to be there. However it would take a fair amount of work to detach it as something that can be just "pulled in". For your personal experimentation, feel free to dig around in the Kernel project, and see how much of it you can stub out. In particular, `Kernel\SystemServices\Threading\ThreadManager.cs` and other bootstrap functions.  
  
I will check on the status of the gitter room.

danwalmsley, 2016-01-06:
@Skobovm hmm after my start code calls _start (not sure where this is defined libMbed?)

it quickly gets stuck in  __malloc_av_ .. PC in RAM somewhere, which is weird because all code runs from flash right?

any ideas how I can get around this?

danwalmsley, 2016-01-06:
Do you also know what provides _sbrk function?

I've been providing my own, but maybe its not compatible with llilum.

Skobovm, 2016-01-06:
Pretty sure it's part of the standard C lib, which should be in the same location as GCC: http://www.gnu.org/software/libc/manual/html_node/Resizing-the-Data-Segment.html