NETMF/llilum #119: The application doesn't start

chutichnuoc, 2016-01-05:
Everything just worked fine before updating "VS update 1"
After installed "VS update 1", we can not debug. The error is shown as below

[image]

After removed all GPIO objects then the application run well. We know it run because we can debug.
Today we updated latest version which included fixing  "VS update 1", and the application does not start at all, although in main function just a while loop as below:

int a = 0;
while (true)
 {
           a++;
          if (a==0xFFFF) 
           a = 0;

    } 
It is always point us to asm("bkpt");
please advice me?

Skobovm, 2016-01-05:
Sorry for the inconvenience, hopefully we can resolve this soon. What board are you currently working with? Have you changed the properties of the project to the correct board? Have you been able to debug the default SDK program? Lastly, if you had this working code before, have you done a clean and rebuild?

chutichnuoc, 2016-01-05:
Enable MbedOSabstraction and Native in Configuration Manager before run CreateDropSDK, now it works.  I am not sure if these steps are necessary. But anyway, it is working. Thank a lot for fixing issue "VS update 1".

Skobovm, 2016-01-05:
Glad you have it working!

closed 2016-01-05 by Skobovm