NETMF/llilum #121: STM32F411, the clock source is 96MHz or 100MHz?

chutichnuoc, 2016-01-06:
I found the file configuration.cs says that it is 100MHz.
But I am not sure in mbed source config, it is 100MHz or 96MHz, because it was compiled, I guess it is 96MHz.
In case we want to change the clock source, is there anything else to change other than 2 these files, please?

Skobovm, 2016-01-06:
Which mbed source config are you referring to? The configuration.cs file doesn't set the frequency, it just allows the system to use that value for timing calculations.

chutichnuoc, 2016-01-06:
Hi, 
We re-compiled mbed source and modified the clock in 2 files:
- stm32f4xx_hal_conf.h 
- system_stm32f4xx.c
Output file should be system_stm32f4xx.o, and we replaced it in the folder:
..\Zelig\mbed\TARGET_NUCLEO_F411RE\TOOLCHAIN_GCC_ARM\system_stm32f4xx.o.

Is there anything else to change to get new clock source others than that file, please?

Skobovm, 2016-01-06:
From the mBed/system point of view, please see this post (and links within): https://developer.mbed.org/users/aung/notebook/system-clock-frequency/
  
Look through the mBed ticker/timer code to see where the SystemCoreClock comes from. If it reads a register, you're all set. If it has a defined macro somewhere, you will have to modify it. I can't recall which it is.

Then modify Configuration.cs to reflect the changes that you made.

chutichnuoc, 2016-01-06:
Thank you too much! I think I got it.

closed 2016-01-06 by chutichnuoc