NETMF/llilum #124: Generate .AXF file?

chutichnuoc, 2016-01-06:
Hi LLilum team,
I have a question, is there any way to generate .axf file to debug directly by Jtag?
if no, are we going to add this feature in future?
Thank a lot!

Skobovm, 2016-01-06:
They're the same thing: https://en.wikipedia.org/wiki/Executable_and_Linkable_Format

Skobovm, 2016-01-06:
Sorry, let me clarify: .axf and .elf are the same thing. Change the extension if you need to, and use it for debugging.

cw2, 2016-01-07:
I can confirm the .elf file renaming to .axf works.

Note: If you happen to use Keil MDK µVision IDE, it sometimes cannot open ELF files produced by GCC (often crashes hard), due to unusual/unsupported section layout. But that is not problem of the file itself or its extension, it is limitation of the Keil IDE...

chutichnuoc, 2016-01-07:
Thanks for useful information. I will try it.

closed 2016-01-07 by chutichnuoc