NETMF/llilum #134: Marking collector corruption or false positive

lt72, 2016-01-11:
REPRO:

running the 'mbed_simple_NET_K64F.FrontEndConfig' seems to repro consistently.
one needs to also run the companion desktop app in the test folder of the solution, at
<repo>\Zelig\Zelig\Test\Desktop\Net\NetworkTest

the address at 0x20008780 seems to fall in themiddle of a thread's  stack, and the flags are 0xFFFFFFFD, which is the return address (EXC_RETURN) for thread mode...

```
Continuing.
DebugLog: IP Address:10.0.1.6
DebugLog: [Last Active Frame] EXC=0xfffffffd, PSR=0x61000000, PC=0x00012062
DebugLog: [Last Active Frame] R0=0x00000003, R1=0xffffffef, R2=0x00000001, R3=0x00000001, R12=0x00000000
DebugLog: Corruption! address=0x20008780, flags=0x000000fd

Program received signal SIGTRAP, Trace/breakpoint trap.
Break () at mbed_debug.cpp:33
33                  asm("bkpt");
=> 0x0001213e <Break()+10>:     00 be   bkpt    0x0000
   0x00012140 <Break()+12>:     70 47   bx      lr
(gdb) bt
#0  Break () at mbed_debug.cpp:33
#1  Breakpoint (n=<optimized out>) at mbed_debug.cpp:50
#2  Microsoft.CortexM3OnMBED.Device::ProcessBugCheck#19814 (code=0x4d000123) at Device.cs:29
#3  Microsoft.Zelig.Runtime.BugCheck::Raise#5477 (code=0x4d000123) at BugCheck.cs:69
#4  Microsoft.Zelig.Runtime.MarkAndSweepCollector::VisitHeapObject#5012 (address=, this=<optimized out>) at MarkAndSweepCollector.cs:850
#5  Microsoft.Zelig.Runtime.MarkAndSweepCollector::VisitInternalPointer#5011 (address=<optimized out>, this=) at MarkAndSweepCollector.cs:764
#6  Microsoft.Zelig.Runtime.ConservativeMarkAndSweepCollector.ConservativeStackWalker::Process#6322 (ctx=<optimized out>, this=<optimized out>) at ConservativeMarkAndSweepCollector.cs:29
#7  Microsoft.Zelig.Runtime.MarkAndSweepCollector::WalkStackFrames#5005 (this=) at MarkAndSweepCollector.cs:528
#8  Microsoft.Zelig.Runtime.ConservativeMarkAndSweepCollector::WalkStackFrames#4964 (this=) at ConservativeMarkAndSweepCollector.cs:80
#9  Microsoft.Zelig.Runtime.MarkAndSweepCollector::StartCollection#5004 (this=) at MarkAndSweepCollector.cs:461
#10 Microsoft.Zelig.Runtime.MarkAndSweepCollector::Collect#4997 (this=) at MarkAndSweepCollector.cs:309
#11 Microsoft.Zelig.Runtime.ConservativeMarkAndSweepCollector::Collect#4962 (this=) at ConservativeMarkAndSweepCollector.cs:57
#12 Microsoft.Zelig.Runtime.DefaultTypeSystemManager::AllocateInner#5812 (vTable=<optimized out>, size=<optimized out>) at DefaultTypeSystemManager.cs:240
#13 Microsoft.Zelig.Runtime.DefaultTypeSystemManager::AllocateObject#5802 (vTable=, this=<optimized out>) at DefaultTypeSystemManager.cs:24
#14 Microsoft.Zelig.LlilumOSAbstraction.CmsisRtos.CmsisRtosMailbox::Create#20669 (queueSize=0x127a5) at CmsisRtosMailbox.cs:22
#15 Microsoft.Zelig.LlilumOSAbstraction.CmsisRtos.CmsisRtos::LLOS_osMessageCreate#20653 (queue_sz=0x4d000123) at CmsisRtos.cs:27
#16 sys_mbox_new (mbox=mbox@entry=, queue_sz=queue_sz@entry=0x8) at sys_arch.c:90
#17 netconn_alloc (t=<optimized out>, callback=callback@entry=<event_callback>) at api_msg.c:608
#18 netconn_new_with_proto_and_callback (t=t@entry=NETCONN_TCP, proto=proto@entry=0x0, callback=callback@entry=<event_callback>) at api_lib.c:73
#19 lwip_socket (domain=<optimized out>, type=<optimized out>, protocol=<optimized out>) at sockets.c:956
#20 LLOS_lwip_socket (family=<optimized out>, type=<optimized out>, protocol=<optimized out>) at mbed_socket.cpp:37
#21 Microsoft.CortexM3OnMBED.HardwareModel.SocketProvider::socket#20055 (family=0x4d000123, type=0x4d000123, protocol=<error reading variable: Cannot access memory at address 0xfffffb60>) at SocketProvider.cs:195
#22 Microsoft.Llilum.Net.SocketNative::socket#4843 (type=0x1, protocol=0x0, family=0x0) at SocketNativeImpl.cs:12
#23 System.Net.Sockets.Socket::.ctor#15111 (addressFamily=0x0, socketType=0x1, protocolType=0x0, this=<optimized out>) at Socket.cs:27
#24 Microsoft.Zelig.Test.mbed.SimpleNet.Program::Main#144 () at Program.cs:51
#25 Microsoft.Zelig.Runtime.ThreadManager::MainThread#3689 () at ThreadManager.cs:557
#26 System.Threading.ThreadStart::Invoke#8785 (this=<optimized out>) at ArrayImpl.cs:262
#27 System.Threading.Thread::Entrypoint#4541 (this=) at ThreadImpl.cs:220
#28 ?? ()
(gdb) i r
r0             0xd      0xd
r1             0xd      0xd
r2             0xfffffb60       0xfffffb60
r3             0x1010001        0x1010001
r4             0x1fff0ad8       0x1fff0ad8
r5             0x20008780       0x20008780
r6             0x20008778       0x20008778
r7             0xfd     0xfd
r8             0x1fff0ad8       0x1fff0ad8
r9             0x1fff0ad8       0x1fff0ad8
r10            0xfffffd50       0xfffffd50
r11            0x20008780       0x20008780
r12            0x1fff0d40       0x1fff0d40
sp             0x2000d1e0
lr             0x12157  0x12157
pc             0x1213e  <Break()+10>
xpsr           0x1000000        0x1000000
msp            0x2002ff58       0x2002ff58
psp            0x2000d1e0       0x2000d1e0
primask        0x0      0x0
control        0x2      0x2
basepri        0x10     0x10
faultmask      0x0      0x0
(gdb) x/16wx 0x20008780
0x20008780:     0x00000008      0x0002ba88      0x200097b0      0x2000985c
0x20008790:     0xffffffff      0xffffffff      0x2000c2d8      0x00000008
0x200087a0:     0x0002ba28      0x00000400      0x00000000      0x00000000
0x200087b0:     0x00000000      0x00000000      0x00000000      0x00000000
(gdb) x/16wx 0x20008780-8
0x20008778:     0xfffffffd      0x00000000      0x00000008      0x0002ba88
0x20008788:     0x200097b0      0x2000985c      0xffffffff      0xffffffff
0x20008798:     0x2000c2d8      0x00000008      0x0002ba28      0x00000400
0x200087a8:     0x00000000      0x00000000      0x00000000      0x00000000
(gdb) x/16wx 0x20008780-16
0x20008770:     0x200097a8      0x20009660      0xfffffffd      0x00000000
0x20008780:     0x00000008      0x0002ba88      0x200097b0      0x2000985c
0x20008790:     0xffffffff      0xffffffff      0x2000c2d8      0x00000008
0x200087a0:     0x0002ba28      0x00000400      0x00000000      0x00000000
```

lt72, 2016-01-15:
I suspect that we are trying here to parse a RegistersOnStack struct that was pushed on the stack

jelin1, 2016-01-20:
Fixed with 52f2dddfedbcfb03c313bf4f6a126e3655d2bcc2

closed 2016-01-20 by jelin1