NETMF/llilum #137: System.Threading.EventWaitHandle.EventWaitableObject::Acquire has a lot of unnecessary null checks

lt72, 2016-01-13:


ColinA-MSFT, 2016-01-13:
We should only null-check once per variable.  However, in this case there are multiple field access operators for the same value, each with different result variables.

We should add an optimization to skip the field access and re-use the original value when possible.

ColinA-MSFT, 2016-01-13:
Upon inspection, m_owner is marked as volatile, so this code is correct.  However, there's no need for m_owner to be volatile when m_state is the only variable that can be modified concurrently.

ColinA-MSFT, 2016-02-03:
It appears these field accesses and null checks come directly from MSIL.  We could probably do more to reduce them, though.  I'm reducing the priority as this is a potential optimization, but not required for correctness.