NETMF/llilum #142: STM32F401 in STM32F091 board project

cw2, 2016-01-18:
I have noticed there are "STM32F401" in namespaces and assembly attributes in the STM32F091 board projects - most likely missed during text replace (?)

Also, should not be ArmV6M architecture used instead (of ArmV7M)?
```
    [Defaults("Platform", typeof(Microsoft.Zelig.Configuration.Environment.Abstractions.Architectures.ArmV7M))]
    [Defaults("CallingConvention", typeof(Microsoft.Zelig.Configuration.Environment.Abstractions.Architectures.ArmV7MCallingConvention))]
```
I could prepare pull request with such changes, going through "Creating a BSP" guide for another Cortex-M0 board...

lt72, 2016-01-19:
that would be great!

cw2, 2016-02-07:
Fixed in 32b1f0586544b9891718cd6e7b26721989b1b06d.

closed 2016-02-07 by cw2