NETMF/llilum #162: TypeConsistencyErrorException

maiorfi, 2016-01-30:
While building default SDK template generated project I get:

 Compiling...
1>
1>       9,271 [   0,043]: Phase: ReduceNumberOfTemporaries                          [types:  4122,  fields:   4821,  methods:  26545]
1>      14,831 [   5,561]: Phase: TransformFinallyBlocksIntoTryBlocks                [types:  4122,  fields:   4821,  methods:  26545]
1>      15,747 [   0,916]: Phase: ApplyClassExtensions                               [types:  4122,  fields:   4821,  methods:  26545]
1>      78,633 [  62,886]: Phase: PrepareImplementationOfInternalMethods             [types:  4039,  fields:   4807,  methods:  26071]
1>  Caught exception: Microsoft.Zelig.TypeConsistencyErrorException: 'AbstractReferenceTypeRepresentation(Microsoft.Zelig.Runtime.GarbageCollectionManager)' is marked as ForcedInstantiation, but found multiple candidates: Microsoft.Llilum.LPC1768.StrictReferenceCountingCollector, Microsoft.Llilum.LPC1768.ReferenceCountingCollector, Microsoft.Llilum.LPC1768.ConservativeMarkAndSweepCollector
1>     in Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildHierarchyTables() in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:riga 1026
1>     in Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildCrossReference() in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:riga 889
1>     in Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Phases.PrepareImplementationOfInternalMethods.Run() in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Phases\PrepareImplementationOfInternalMethods.cs:riga 38
1>     in Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.PhaseDriver.Execute() in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\PhaseDrivers\PhaseDriver.cs:riga 124
1>     in Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Controller.ExecuteSteps(Boolean fGenerateImageOnly) in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Controller.cs:riga 293
1>     in Microsoft.Zelig.FrontEnd.Bench.Compile() in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:riga 1853
1>     in Microsoft.Zelig.FrontEnd.Bench.RunBench(String[] args) in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:riga 2310
1>     in Microsoft.Zelig.FrontEnd.Bench.<>c__DisplayClass127_0.<Main>b__0() in C:\llilum\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:riga 2325

What's actually wrong?

Thanks!

maiorfi, 2016-01-30:
Ok, solved specifying

-CompilationOption System.String GarbageCollectionManager ConservativeMarkAndSweepCollector

in .FrontEndConfig

miloush, 2016-02-03:
Same here. The workaround works, but maybe the issue should stay open until the SDK template works out of the box?

maiorfi, 2016-02-03:
Right :smile:

lt72, 2016-02-09:
Fixed with 7eac0008422414e779fd4642802d37155bec4c36

closed 2016-02-09 by lt72