NETMF/llilum #163: Incremental build of user application?

maiorfi, 2016-01-30:
What is the proper way to update and redeploy a user (i.e. managed) application based on SDK template?

It seems that at this time full solution rebuild is needed, that is quite slow :cry: 

Thanks!

piwi1263, 2016-01-30:
FWIW: I have to do a complete solution rebuild as well. And that takes it time. I do only, well almost, change the managed part but sofar, the only thing I've got to work is the blinky thing (on a Nucleo-F411RE), tried I2C before the blinky code, again managed, did compile and deploy but no blinky so I reckon some break up .... and with no simple Debug.Print ... this is somewhat getting used to ...  

Q: Did you get managed serial to work on the Nucleo's arduino header pins, or any pins at all ?

josesimoes, 2016-02-01:
Same here: several minutes for a rebuid, no matter what is changed. Is this the expected behaviour? Is there an optimization/options that is still missing from the documentation or project properties?

lt72, 2016-02-01:
I hear you!
and I am working on it.

Doing AOT means that we build a type system, and then we modify the items inside it (types, methods)
Doing incremental build is a big deal... 

I am working on a solution but it will take some more time, in the order of a 8 weeks at least.

maiorfi, 2016-02-02:
Hi Lorenzo.

It's definitely great to hear that you are working for that goal.

Just for sake of curiosity: does this feature actually require 8 weeks or is it subordinated to other higher-priority features already in the roadmap?

Also, couldn't using netmf-interpreter (despite it provides a less-rich C# support) be a way to write and debug user application before building release version through Llilum toolchain?

lt72, 2016-02-09:
Hi Lorenzo,
yes, you are correct. It would be a good way to proceed. Which is also part of the reason why I started to port the NETMF assemblies over to LLILUM. Unfortunately we are at a stage where we are having a right time to pick the right direction for the project. NETMF full compat will take around 4 weeks to fully test. I am considering skipping that altogether and going for LLILUM incremental build directly.  

What are your thoughts/suggestions?

maiorfi, 2016-02-09:
Incremental build is no doubt more compelling: in 10 weeks we'll have much more user (i.e. pseudo-managed) code written and tested :smile: