NETMF/llilum #168: SerialDevice.FromIdAsync() return type question

piwi1263, 2016-02-01:
Is the return type correct here for the SerialDevice.FromIdAsync() ?

[image]

Or should this match more to return just the Device as is done by the I2cDevice and the SpiDevice ?

Skobovm, 2016-02-02:
Hi,  
  
Sorry about the confusion there. The SerialDevice.FromIdAsync is the method that actually has the correct return type, while the ones for SpiDevice and I2cDevice are incorrect.

piwi1263, 2016-02-02:
At least it is cleared, but using the await for the SerialDevice.FindIdAsync() returns into a GetAwaiter not implemented/not found error.

Skobovm, 2016-02-02:
For now, please use .GetResults() on the IAsyncOperation. You can also use System.IO.Ports.SerialPort for serial communication. For all other peripherals, I would recommend using the classes in Microsoft.Llilum.Devices, as they are more lightweight than their UWP equivalents

piwi1263, 2016-02-02:
Hi Mikhail,

thanks for answer, I'll give it a shot and see how far I come.

Cheers,

Peter.

closed 2016-02-25 by piwi1263