NETMF/llilum #171: Building LibLLVM

miloush, 2016-02-03:
Hi,

I tried to do the first build as described on the wiki: https://github.com/NETMF/llilum/wiki/setup

After opening the Zelig solution and trying to build it, Llvm.NET complains first:
`LLVM_SRCROOT_DIR not specified, either set it manually in  your project file or set the registry value HKLM\Software\LLVM\SrcRoot`

The registry suggestion does not work, but setting the environment variable does. Doing that, I can build all projects except LibLLVM:

[image]

I am not sure how to fix this though, I built the LLVM 3.7.0 linked from the wiki page using the BuildLlvmWithVS.cmd script file, CMake 3.4.3, Python 2.7.10 and ARM GCC 5-2015-q4-major.

Has anyone else seen this?

miloush, 2016-02-03:
I have just noticed https://github.com/NETMF/llilum/commit/41c16cf151358adeb77c347718811158c8033dd9 so I am building 3.8.0 now. If that helps, it might be worth updating the wiki.

miloush, 2016-02-03:
I confirm I was able to successfully build the Zelig solution as instructed with LLVM 3.8.0 RC1, though I still had to set the LLVM_SRCROOT_DIR.

microcodeman, 2016-02-03:
I am seeing similar errors too. I am trying to build Zelig from the command line after setting the environment variable for  LLVM_SRCROOT_DIR.

I would appreciate knowing your attempt with LLVM 3.8.0

microcodeman, 2016-02-03:
Awesome Miloush, will give it a go tonight. How long did LLVM 3.8.0 take to build ?

miloush, 2016-02-03:
Forever.

miloush, 2016-02-03:
I don't think I recorded the start time but 6-8 hours seems to be reasonable estimate. Doing it overnight makes sense.

smaillet-ms, 2016-02-03:
The WIKI was updated (My apologies on that I intended to do it immediately after pushing that change but I was interrupted and forgot to get back to that until I came in this morning)

6 hours seems to be a fairly common average build time. We are looking into publishing LLVM.NET on it's own repo and using NUGET to pull it into Llilum. That way everyone doesn't need to build LLVM itself all the time.

cw2, 2016-02-03:
I think it should be possible to reduce LLVM build time by decreasing the number of supported platforms, depending on what exactly is needed for Llilum (x86_64, arm (?)). If I remember it correctly, I made ARM-only LLVM build for another project some time ago, but unfortunately I have not measured the build time... I guess I can try to play with it now, for 3.8...

smaillet-ms, 2016-02-03:
The targets aren't that big an impact - it's the x64 vs 32 bit full builds across release, debug and MinRelWithDebInfo that's the big time killer. Packaging via NUGET simplifies usage for everyone and keeps LLVM.NET re-usable outside of Llilum. (e.g. LLVM.NET was designed as a general LLVM runtime binding for .NET, not just for Llilum)