NETMF/llilum #172: SecurityException in SDK template

miloush, 2016-02-03:
Might be related to #53.

When trying to debug the SDK template, the controller.OpenPin call ends up with 
```
System.Security.SecurityException was unhandled
  HResult=-2146233078
  Message=ECall methods must be packaged into a system module.
  Source=Llilum.Devices
  StackTrace:
       at Microsoft.Llilum.Devices.Gpio.GpioPin.TryAcquireGpioPin(Int32 pinNumber)
       at Microsoft.Llilum.Devices.Gpio.GpioPin.TryCreateGpioPin(Int32 pinNumber) in C:\src\llilum\Zelig\Zelig\RunTime\Framework\Llilum\Devices\Gpio\GpioPin.cs:line 105
       at Windows.Devices.Gpio.DefaultPinProvider..ctor(Int32 pinNumber) in C:\src\llilum\Zelig\Zelig\RunTime\Framework\Windows\Devices\Gpio\DefaultPinProvider.cs:line 20
       at Windows.Devices.Gpio.GpioController.TryOpenPin(Int32 pinNumber, GpioSharingMode sharingMode, GpioPin& pin, GpioOpenStatus& status) in C:\src\llilum\Zelig\Zelig\RunTime\Framework\Windows\Devices\Gpio\GpioController.cs:line 98
       at Windows.Devices.Gpio.GpioController.OpenPin(Int32 pinNumber, GpioSharingMode sharingMode) in C:\src\llilum\Zelig\Zelig\RunTime\Framework\Windows\Devices\Gpio\GpioController.cs:line 66
       at Windows.Devices.Gpio.GpioController.OpenPin(Int32 pinNumber) in C:\src\llilum\Zelig\Zelig\RunTime\Framework\Windows\Devices\Gpio\GpioController.cs:line 49
       at Managed.Program.Main() in C:\src\llilum\projects\LlilumApplication1\LlilumApplication1\Managed\Program.cs:line 60
```

Is there anything I can do to get past this?
Thanks!

lt72, 2016-02-09:
Mikhail,

could you give a look to this one if at all possible?

Skobovm, 2016-02-09:
Hmm, I'm not really sure where that would come from... Can you tell me what board you are using and what pin you are trying to open?

Do you get that issue when debugging the board using the SDK, or are you debugging the board from within a SDK debug session?

miloush, 2016-02-10:
Hi, it is a mbed 1768 board, I just followed https://github.com/NETMF/llilum/wiki/SDK-User-Guide (and #162 workaround), i.e. created project using the VS template and started debugging in the Visual Studio.

Skobovm, 2016-02-10:
I have tried to repro this issue several ways but have had no luck getting that exception. The SDK will be updated very soon, and I will notify you when that happens. When it does, please re-install the SDK, and try to build the default application, as it has been working fine for the past few months.

lt72, 2016-03-27:
no repro

closed 2016-03-27 by lt72