NETMF/llilum #178: Error builing default project

microcodeman, 2016-02-05:
Caught exception: Microsoft.Zelig.TypeConsistencyErrorException: 'AbstractReferenceTypeRepresentation(Microsoft.Zelig.Runtime.GarbageCollectionManager)' is marked as ForcedInstantiation, but found multiple candidates: Microsoft.Llilum.LPC1768.StrictReferenceCountingCollector, Microsoft.Llilum.LPC1768.ReferenceCountingCollector, Microsoft.Llilum.LPC1768.ConservativeMarkAndSweepCollector
1>     at Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildHierarchyTables() in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:line 1026
1>     at Microsoft.Zelig.CodeGeneration.IR.TypeSystemForCodeTransformation.BuildCrossReference() in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\TypeSystemForCodeTransformation.cs:line 889
1>     at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Phases.PrepareImplementationOfInternalMethods.Run() in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Phases\PrepareImplementationOfInternalMethods.cs:line 38
1>     at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.PhaseDriver.Execute() in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\PhaseDrivers\PhaseDriver.cs:line 124
1>     at Microsoft.Zelig.CodeGeneration.IR.CompilationSteps.Controller.ExecuteSteps(Boolean fGenerateImageOnly) in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\CodeTransformation\CompilationSteps\Controller.cs:line 293
1>     at Microsoft.Zelig.FrontEnd.Bench.Compile() in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 1855
1>     at Microsoft.Zelig.FrontEnd.Bench.RunBench(String[] args) in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 2312
1>     at Microsoft.Zelig.FrontEnd.Bench.<>c__DisplayClass128_0.<Main>b__0() in C:\src\llilum\Zelig\Zelig\CompileTime\CodeGenerator\FrontEnd\Bench.cs:line 2327

microcodeman, 2016-02-05:
The above continues to 

```
Compiling and optimizing with LLVM: ARM\Debug\Llilum.Managed.bc to ARM\Debug\Llilum.Managed.o
1>  c:\src\llvm-3.8.0rc1.src.tar\llvm-3.8.0rc1.src\llvm-3.8.0rc1.src\build\x64\Debug\bin\opt: ARM\Debug\Llilum.Managed.bc: error: Could not open input file: no such file or directory
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Application Type\Llilum\1.0\Android.Common.targets(374,5): error MSB3073: The command "c:\src\llvm-3.8.0rc1.src.tar\llvm-3.8.0rc1.src\llvm-3.8.0rc1.src\build\x64\Debug\bin\opt -scalarrepl -targetlibinfo -verify -mem2reg -constmerge -adce -globaldce -time-passes ARM\Debug\Llilum.Managed.bc -o ARM\Debug\Llilum.Managed_opt.bc " exited with code 1.

```

Skobovm, 2016-02-05:
Hi,
  
I noticed this issue a few days ago - it is a very simple fix. Please open `<repo root>\Zelig\BoardConfigurations\LPC1768\Board\SystemServices\GarbageCollectionManager.cs` and comment out the ReferenceCountingCollector and the StrictReferenceCountingCollector. Then rebuild the solution, and copy the LPC1768.dll file from `..\..\..\ZeligBuild\Target\bin\Debug\` to where your SDK is located (`<SDKRoot>\ZeligBuild\Target\bin\Debug\`).

microcodeman, 2016-02-05:
Thank you that worked.

I also notice that there has been an issue and FrontEndConfiguration solution for this.

#

closed 2016-02-05 by microcodeman