NETMF/llilum #182: SDK build script missing?

josesimoes, 2016-02-15:
After a successful testing a few weeks ago, I was going through the process of updating the repository and rebuilding LLVM and the SKD.
The Wiki instructions at SDK-User-Guide require the build of the SDK using a script from the scripts file.
It seems that this script is missing. I've searched for it and found that commit 560f531 removed it.
Is the script being reworked or the wiki instructions need to be updated?

Thanks!

Skobovm, 2016-02-16:
The wiki needs updating. @smaillet-ms could you please update the wiki to the most recent steps to follow after your MSI changes?

@josesimoes The SDK install process includes running an MSI now. Try following these steps:
1. Open Visual Studio, go to Tools -> Extensions and Updates, and delete the Llilum Application extension
2. Do a clean pull of the repo
3. Open System environment variables and delete any that have the word LLILUM in them
4. Open the Zelig solution and build it
5. Open and build the solution `<repo root>\LlilumSDK\LlilumSDK.sln`
6. There should now be an MSI file under `<repo root>\Zelig\ZeligBuild\Host\bin\Debug\LlilumSDK.msi`
7. Run that MSI
8. Restart so that environment variable changes are picked up
9. Open and build the solution `<repo root>\VisualStudio\LlilumProjectType\LlilumApplication.sln`
10. Run the VSIX installer that was created `<repo root>\VisualStudio\LlilumProjectType\LlilumApplication\LlilumApplication.ProjectType\bin\Debug\LlilumApplication.vsix`  
  
These steps should unblock you. We will have a consolidated wiki with similar steps shortly - sorry for the confusion

josesimoes, 2016-02-16:
@Skobovm thanks for the update and detailed explanations.
I've followed your steps and with the help of the SDK readme successfully build the SDK.

closed 2016-02-16 by josesimoes