NETMF/llilum #185: Support for x86, llilum SDK

BitProgress, 2016-02-20:
I am trying to build the full development system using win10 32-bit.

Does the "llilum SDK" installer project not support x86?

Should I stop trying to build on 32-bit due to limitations on the current state of the llilum development? Or should it be possible to use 32-bit OS?

Thanks,
Martin Jørgensen (martin@bitprogress.com)

BitProgress, 2016-02-21:
When I try to build the SDK on a 64-bit machine, I also get errors regarding the special characters in:
\Zelig\tools\openocd\scripts\target\1986ве1т.cfg
\Zelig\tools\openocd\scripts\target\к1879xб1я.cfg

The project "LlilumSDK" is supposed to be build, right? In the solution, it is not enabled for build.

Skobovm, 2016-02-22:
Hi Martin,
  
Please work with the 64 bit OS, 32 bit is not supported.  
  
In order to get set up with the SDK, do the following:
1. If you're blocked by something in openocd, delete everything in that folder (you can get it back later)
2. Follow the setup steps here: https://github.com/NETMF/llilum/wiki/setup
3. Get everything from the prerequisites section mentioned here: https://github.com/NETMF/llilum/wiki/SDK-User-Guide
4. Follow the steps from my comment in this issue: https://github.com/NETMF/llilum/issues/182