NETMF/llilum #186: Build as Release not possible

BitProgress, 2016-02-21:
Dear developer team,

I know you are very busy developing on this interesting project, and I look forward to the result!

I have tried to build some of the solutions as release (with optimization etc.), but it is not possible. I have also tries to build the default Llilum application (for the LPC1768) in release, which also didn't work.

Actually I am very interested in the amount of memory (Flash) needed for the smallest application. When I look into the list file for the debug build, a lot of (unused?) methods are compiled.

Best regards
Martin

lt72, 2016-03-01:
Hi BitProgress.

the minimum size app (<repo>\llilum\Zelig\Zelig\Test\mbed\simple\Program.cs) will take approximately ~170KB of flash space and use 32KB of RAM. This is a fully multi-threaded app.

We will enable shortly a single threaded app (classic loop + ISR approach) that should run in about ~80KB and use <16KB RAM.

The system build all the code that is in the solution, but then drops all the code that is not reachable.
If you perform a build from the solution file (<repo>\llilum\Zelig\Zelig\Zelig.sln), you will see that we start with the following type system: 

```types:  4200,  fields:   4910,  methods:  27097```

but we finish with 

```types:   364,  fields:    678,  methods:   1593```

all unused types/methods/fields are dropped during compilation.

BitProgress, 2016-03-01:
Hi Lorenzo,

I really appreciate you took the time to explain. I just got the mbed boards, and looking forward to see how the debug experience is :-)

Looking forward for more interesting information regarding llilium.

BR Martin

smaillet-ms, 2016-03-02:
Can you provide detailed step by step instruction to reproduce the problem you encountered?

BitProgress, 2016-03-06:
To reproduce the problem simply create a new llilum application in visual studio, and change build configuration to: "Release".

It gives the following error:

`
1>------ Build started: Project: Native, Configuration: Release ARM ------
2>------ Build started: Project: Managed, Configuration: Release Any CPU ------
1>  mbed_asm.S
2>  Managed -> D:\Llilum\Projects\LlilumApplicationTestRelease\LlilumApplicationTestRelease\Managed\bin\Release\Llilum.Managed.exe
1>  startup_LPC17xx.S
1>  pch.h
1>  pch.h
1>  Source.cpp
1>  mbed_adc.cpp
1>  mbed_clock.cpp
1>  mbed_core.cpp
1>  mbed_debug.cpp
1>  mbed_ethernet.cpp
1>  In file included from C:\Program Files\Microsoft\Llilum\mbed/Serial.h:23:0,
1>                   from C:\Program Files\Microsoft\Llilum\mbed/mbed.h:44,
1>                   from C:\Program Files\Microsoft\Llilum\os_layer\ports\mbed\mbed_ethernet.cpp:6:
1>  C:\Program Files\Microsoft\Llilum\mbed/Stream.h:34:19: error: macro "putc" requires 2 arguments, but only 1 given
1>       int putc(int c);
1>                     ^
1>  In file included from C:\Program Files\Microsoft\Llilum\mbed/mbed.h:51:0,
1>                   from C:\Program Files\Microsoft\Llilum\os_layer\ports\mbed\mbed_ethernet.cpp:6:
1>  C:\Program Files\Microsoft\Llilum\mbed/RawSerial.h:67:19: error: macro "putc" requires 2 arguments, but only 1 given
1>       int putc(int c);
1>                     ^
1>  In file included from C:\Program Files\Microsoft\Llilum\mbed/platform.h:26:0,
1>                   from C:\Program Files\Microsoft\Llilum\mbed/mbed.h:21,
1>                   from C:\Program Files\Microsoft\Llilum\os_layer\ports\mbed\mbed_ethernet.cpp:6:
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:88:9: error: '::random' has not been declared
1>   using ::random;
1>           ^
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:89:9: error: '::srandom' has not been declared
1>   using ::srandom;
1>           ^
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:96:9: error: '::unlockpt' has not been declared
1>   using ::unlockpt;
1>           ^
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:97:9: error: '::ptsname' has not been declared
1>   using ::ptsname;
1>           ^
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:98:9: error: '::ptsname_r' has not been declared
1>   using ::ptsname_r;
1>           ^
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:99:9: error: '::getpt' has not been declared
1>   using ::getpt;
1>           ^
1>  c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdlib:100:9: error: '::grantpt' has not been declared
1>   using ::grantpt;
1>           ^
1>  In file included from c:\programdata\microsoft\androidndk\android-ndk-r10e\sources\cxx-stl\system\include\cstdio:37:0,
1>                   from C:\Program Files\Microsoft\Llilum\mbed/platform.h:27,
1>                   from C:\Program Files\Microsoft\Llilum\mbed/mbed.h:21,
1>                   from C:\Program Files\Microsoft\Llilum\os_layer\ports\mbed\mbed_ethernet.cpp:6:
1>  C:\Program Files\Microsoft\Llilum\mbed/Stream.h:36:9: error: expected unqualified-id before '--' token
1>       int getc();
1>           ^
1>  C:\Program Files\Microsoft\Llilum\mbed/Stream.h:36:9: error: expected ')' before '--' token
1>  C:\Program Files\Microsoft\Llilum\mbed/RawSerial.h:73:9: error: expected unqualified-id before '--' token
1>       int getc();
1>           ^
1>  C:\Program Files\Microsoft\Llilum\mbed/RawSerial.h:73:9: error: expected ')' before '--' token
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Application Type\Llilum\1.0\Android.Common.targets(218,5): error MSB6006: "arm-none-eabi-gcc.exe" exited with code 1.
========== Build: 1 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
`

lt72, 2016-09-01:
General issues with release builds are now fixed with 913fdc4b262739c36b88d78ce9acfbda4ca346d9. 
Will check SDK status soon.